coolsnowwolf/lede #6220: 能否更新一下docker

star-o0O, 2021-01-19:
固件里带的docker-ce是19.03.12，官方最新release是20.10.2。看发行说明似乎修复了许多莫名其妙的问题，以及性能改进。
目前docker-ce应该是改名为docker了，这样就没法通过openwrt软件源直接更新。
尝试对照openwrt官方源码修改L大源码中docker相关文件，发现有许多patch以及依赖，技术不精无从下手。
请问能不能更新一下？

NagaseKouichi, 2021-01-19:
端口暴露在WAN口的问题修复了吗？

xiaobaishu1, 2021-01-20:
另一个地方就说让降级回18.06 说新版有问题 这里就说新版有提升性能 能正常使用就想着花里胡哨

star-o0O, 2021-01-20:
> 端口暴露在WAN口的问题修复了吗？

L大这个不是可选禁止wan口访问docker么

NagaseKouichi, 2021-01-20:
> > 端口暴露在WAN口的问题修复了吗？
> 
> L大这个不是可选禁止wan口访问docker么

不知道怎么回事，我没有打开WAN口访问，但是测试发现WAN口依然可以访问。
我现在改成先在op用LXC装debian，然后再在debian下装docker了

sxx1314, 2021-01-20:
你说的是moby吧，moby社区维护目前最高版本20.10.2
docker-ce,docker官方维护,内含engine(moby) cli packaging,目前最高版本仅19.03.14
如果想要moby推荐直接使用openwrt官方的packages,目前直接就是最高版本。

sxx1314, 2021-01-20:
另外docker-ce 的master其实已经是20.10.2 了，只不过docker官方的人不知道是谨慎还是懒没发布Releases

hxlls, 2021-01-20:
[image]
好像是更换了地址，换成了moby的了，原有的已经抛弃？

sxx1314, 2021-01-20:
> [image]
> 好像是更换了地址，换成了moby的了，原有的已经抛弃？

虽然说不维护了但是依旧在更新。
不清楚状况  推荐直接用 openwrt官方的packages即可。

hxlls, 2021-01-20:
我看了下 openwrt官方的packages的编译文件，好像就是HASH和版本号改了。。
[image]

sxx1314, 2021-01-20:
> 我看了下 openwrt官方的packages的编译文件，好像就是HASH和版本号改了。。地址没变，可能是转跳了
> [image]
仔细看 这是cli 不是完整的docker-ce的组件，还有一个插件内才有engine(moby) 。两个结合才能用

star-o0O, 2021-01-20:
> 你说的是moby吧，moby社区维护目前最高版本20.10.2
> docker-ce,docker官方维护,内含engine(moby) cli packaging,目前最高版本仅19.03.14
> 如果想要moby推荐直接使用openwrt官方的packages,目前直接就是最高版本。

大佬明白人，对于docker的各种分支我已经懵了。
这两天也看了很多，有说docker-ce随docker发布Releases，但瞄了一眼，确实没有。
前两天看openwrt官方的packages中没有docker-ce，还以为是改名了。后来仔细看发现是把docker-ce拆成cli和engine两部分（应该）。
目前理解packages中docker指命令行工具，dockerd指容器引擎，两个要合起来用。
基于这个理解，把docker和dockerd丢进去编译倒是可以用了。

closed 2021-01-20 by star-o0O

sxx1314, 2021-01-20:
[image]

643406834, 2021-04-08:
> > 你说的是moby吧，moby社区维护当前最高版本20.10.2 docker 
> > -ce，docker官方维护，内含引擎（moby）cli包装，目前最高版本仅19.03.14
> > 如果想要moby推荐直接使用openwrt官方的包，当前直接就是最高版本。
> 
> 大佬明白人，对于搬运工的各种分支我已经懵了。
> 这两天也看了很多，有说搬运工-CE随泊坞窗发布新闻稿，但瞄了一眼，确实没有。
> 前两天看的OpenWrt官方的包中没有搬运工-CE，还以为是改名了。后来仔细看发现是把码头工人策拆成CLI和发动机两部分（应该）。
> 目前理解的包中泊坞窗指命令行工具，dockerd指容器引擎，两个要合起来用。
> 基于这个理解，把docker和dockerd丢进去编译倒是可以用了。

docker和dockerd的下载链接发下，正在学习下编译docker，官方默认编译后不能运行