coolsnowwolf/lede #6223: 歌华链内核报错 [PMF]PMF_PerformRxFrameAction: ERROR FRAME

Ivanriver, 2021-01-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
没有类似的issue

## 2. 详细叙述

路由功能正常，但是内核日志大量报错[PMF]PMF_PerformRxFrameAction: ERROR FRAME

### (2) 路由器型号和固件版本

型号：歌华链GeHua GHL-R-001
固件版本：OpenWrt R21.1.18 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

[    0.000000] Linux version 5.4.86 (ccjx@Debian) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3023-3fedff661)) #0 SMP Mon Jan 18 07:53:45 2021
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is GeHua GHL-R-001
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
[    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00038f80
[    0.000000] Readback ErrCtl register=00038f80
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509864K/524288K available (5730K kernel code, 249K rwdata, 1240K rodata, 1280K init, 238K bss, 14424K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015505] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
[    0.055861] pid_max: default: 32768 minimum: 301
[    0.065226] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.079634] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.098095] rcu: Hierarchical SRCU implementation.
[    0.108307] smp: Bringing up secondary CPUs ...
[    6.348635] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.348645] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.348660] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... 900 lines cut ...]
[ 1029.661916] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1031.680040] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1033.695866] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1035.719367] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1044.620232] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1046.651869] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1048.670218] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1050.685144] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1052.702934] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1054.721630] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1056.731353] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1065.657017] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1067.688861] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1069.691139] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1071.707343] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1073.725799] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1075.752970] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1077.755301] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1086.628185] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1088.650148] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1090.651251] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1092.680805] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1094.686229] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1096.698850] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[ 1098.715144] [PMF]PMF_PerformRxFrameAction: ERROR FRAME

AmadeusGhost, 2021-01-22:
看样子是 7603 驱动的报错

Ivanriver, 2021-01-25:
> 看样子是 7603 驱动的报错

关闭无线的MFPC之后，没有再报错了

closed 2021-01-25 by Ivanriver