coolsnowwolf/lede #6224: 编译x86 固件报如下错误 bash: /home/adam/lede/staging_dir/host/bin/python: No such file or directory

adamgonglj, 2021-01-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
install -d -m0755 /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/usr/bin
install -m0755 /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/zerotier-one /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/usr/bin/
ln -sf zerotier-one /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/usr/bin/zerotier-cli
ln -sf zerotier-one /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/usr/bin/zerotier-idtool
cp -fpR ./files/* /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/
find /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/adam/lede/staging_dir/host/bin/sstrip" STRIP_KMOD="/home/adam/lede/scripts/strip-kmod.sh" PATCHELF="/home/adam/lede/staging_dir/host/bin/patchelf" /home/adam/lede/scripts/rstrip.sh /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier
rstrip.sh: /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/usr/bin/zerotier-one: executable
(cd /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_zerotier_conffiles" > conffiles;  )
install -d -m0755 /home/adam/lede/bin/packages/x86_64/packages
/home/adam/lede/scripts/ipkg-build -c -o 0 -g 0 /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier /home/adam/lede/bin/packages/x86_64/packages
Packaged contents of /home/adam/lede/build_dir/target-x86_64_musl/ZeroTierOne-1.6.2/ipkg-x86_64/zerotier into /home/adam/lede/bin/packages/x86_64/packages/zerotier_1.6.2-2_x86_64.ipk
echo "zerotier" >> /home/adam/lede/staging_dir/target-x86_64_musl/pkginfo/zerotier.default.install
make[3]: Leaving directory '/home/adam/lede/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#101.55#8.92#109.39
make[3]: Entering directory '/home/adam/lede/feeds/packages/devel/ninja'
make[3]: Leaving directory '/home/adam/lede/feeds/packages/devel/ninja'
time: package/feeds/packages/ninja/host-compile#0.18#0.05#0.22
make[3]: Entering directory '/home/adam/lede/feeds/packages/devel/meson'
make[3]: Leaving directory '/home/adam/lede/feeds/packages/devel/meson'
time: package/feeds/packages/meson/host-compile#0.15#0.05#0.19
make[3]: Entering directory '/home/adam/lede/feeds/packages/utils/zstd'
rm -f /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/.configured_*
rm -f /home/adam/lede/staging_dir/target-x86_64_musl/stamp/.zstd_installed
/home/adam/lede/staging_dir/host/bin/sed -e "s|@CC@|'gcc',|" -e "s|@CXX@|'g++',|" -e "s|@PKGCONFIG@|/home/adam/lede/staging_dir/host/bin/pkg-config|" -e "s|@CFLAGS@|'-O2', '-I/home/adam/lede/staging_dir/host/include', '-I/home/adam/lede/staging_dir/hostpkg/include', '-I/home/adam/lede/staging_dir/target-x86_64_musl/host/include', '-I/home/adam/lede/staging_dir/host/include', '-I/home/adam/lede/staging_dir/hostpkg/include', '-I/home/adam/lede/staging_dir/target-x86_64_musl/host/include',|" -e "s|@CXXFLAGS@|'-I/home/adam/lede/staging_dir/host/include', '-I/home/adam/lede/staging_dir/hostpkg/include', '-I/home/adam/lede/staging_dir/target-x86_64_musl/host/include',|" -e "s|@LDFLAGS@|'-L/home/adam/lede/staging_dir/host/lib', '-L/home/adam/lede/staging_dir/hostpkg/lib', '-L/home/adam/lede/staging_dir/target-x86_64_musl/host/lib',|" -e "s|@PREFIX@|/home/adam/lede/staging_dir/hostpkg|" < /home/adam/lede/staging_dir/hostpkg/lib/meson/openwrt-native.txt.in > /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/openwrt-native.txt
/home/adam/lede/staging_dir/host/bin/sed -e "s|@CC@|'x86_64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'x86_64-openwrt-linux-musl-g++',|" -e "s|@AR@|x86_64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|x86_64-openwrt-linux-musl-strip|" -e "s|@NM@|x86_64-openwrt-linux-musl-gcc-nm|" -e "s|@LD@|x86_64-openwrt-linux-musl-ld|" -e "s|@PKGCONFIG@|/home/adam/lede/staging_dir/host/bin/pkg-config|" -e "s|@CFLAGS@|'-pipe', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-result', '-ffile-prefix-map=/home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5=zstd-1.4.5', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-O3', '-I/home/adam/lede/staging_dir/target-x86_64_musl/usr/include', '-I/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include', '-I/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify', '-I/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include',|" -e "s|@CXXFLAGS@|'-pipe', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-result', '-ffile-prefix-map=/home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5=zstd-1.4.5', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-O3', '-I/home/adam/lede/staging_dir/target-x86_64_musl/usr/include', '-I/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include', '-I/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify', '-I/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include',|" -e "s|@LDFLAGS@|'-L/home/adam/lede/staging_dir/target-x86_64_musl/usr/lib', '-L/home/adam/lede/staging_dir/target-x86_64_musl/lib', '-L/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib', '-L/home/adam/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib', '-znow', '-zrelro',|" -e "s|@ARCH@|x86_64|" -e "s|@CPU@|"64"|" -e "s|@ENDIAN@|little|" < /home/adam/lede/staging_dir/hostpkg/lib/meson/openwrt-cross.txt.in > /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/openwrt-cross.txt
/home/adam/lede/staging_dir/host/bin/python /home/adam/lede/staging_dir/hostpkg/lib/meson/meson.py  --buildtype plain --native-file /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/openwrt-native.txt --cross-file /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/openwrt-cross.txt  -Dlegacy_level=1 -Ddebug_level=0 -Dbacktrace=false -Dstatic_runtime=false -Dbin_programs=true -Dbin_tests=false -Dbin_control=false -Dzlib=disabled -Dlzma=disabled -Dlz4=disabled -Db_lto=true /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/build/meson/openwrt-build /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/build/meson/openwrt-build/..
bash: /home/adam/lede/staging_dir/host/bin/python: No such file or directory
make[3]: *** [Makefile:107: /home/adam/lede/build_dir/target-x86_64_musl/zstd-1.4.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 127
make[3]: Leaving directory '/home/adam/lede/feeds/packages/utils/zstd'
time: package/feeds/packages/zstd/compile#0.11#0.06#0.17
make[2]: *** [package/Makefile:113: package/feeds/packages/zstd/compile] Error 2
make[2]: Leaving directory '/home/adam/lede'
make[1]: *** [package/Makefile:107: /home/adam/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/adam/lede'
make: *** [/home/adam/lede/include/toplevel.mk:224: world] Error 2
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

daofeng2015, 2021-01-21:
好像是没识别到python环境，有个文件没找到

minlang112, 2021-01-21:
/home/adam/lede/staging_dir/host/bin/python

进到这个目录，查看python这个软链接，对应你系统环境里的文件是否正确，如果不存在的话，自己修改一下。我出现过这个问题，ubuntu19升级到20.04之后，系统的Python变成3.8版本了，但是源码库里的软链接写的是3.7，改一下就好了

adamgonglj, 2021-01-22:
不太会改软连接。

minlang112, 2021-01-23:
> 不太会改软连接。

[image]

adamgonglj, 2021-01-26:
OK，好的。感谢感谢。

closed 2021-01-26 by adamgonglj

dhbowen1, 2021-04-30:
帮了大忙，真TMD，果然Ubuntu升级了下出现这个问题，原来python的问题！

xiaofengcod, 2021-08-09:
win10 子系统咋改啊