coolsnowwolf/lede #6226: 关于用 XRAY 替代 Trojan-Go 和 Socks5 Relay  Redirect（Redsocks2）的讨论

coolsnowwolf, 2021-01-21:
最近 XRAY 的更新更好的支持了 Trojan 模式，包括 UDP Fullcone，也更好的支持了 WS

由于 16M ROM 体积限制，如果同时集成 XRAY 和 Trojan-Go 必然会炸

是否应该 关于用 XRAY 替代 Trojan-Go 和 Socks5 Relay Redirect（Redsocks2）呢？

zh15933, 2021-01-21:
替代吧

accors, 2021-01-21:
最好替代

yushum, 2021-01-21:
你问我支持不支持，我说支持。我就明确的告诉你这一点。

lhy0838, 2021-01-21:
Xray

wutaoll, 2021-01-21:
支持替代！

gjy316, 2021-01-21:
必须替代

coolsnowwolf, 2021-01-21:
顺便替换 SS AEAD 如何？

alienaltman, 2021-01-21:
放16M一条生路吧

68cs, 2021-01-21:
一定要替代

feiyue789, 2021-01-21:
支持替代

SilverHunterLib, 2021-01-21:
支持替代

zgc, 2021-01-21:
代替掉吧

kingdapang, 2021-01-21:
支持替代！trojan-go已经很久没有更新了，xray还支持trojan xtls 更多黑科技值得拥有！

delury, 2021-01-21:
替代吧。

frely, 2021-01-21:
打酱油，不过16M确实可以舍弃了。

waterun, 2021-01-21:
xray

houpi, 2021-01-21:
Xray

MoeGrid, 2021-01-21:
支持替代

KC9226, 2021-01-21:
xray

pegguo, 2021-01-21:
trojan go is better!

AkinoKaede, 2021-01-21:
目前有开发者在为 Xray 添加 SMux，待此项工作完成后，Xray 应该能支持 Trojan-Go 的所有功能。
话说有没有考虑过更好地利用 V2Ray (Xray) 中的路由功能。

PaulYoung0265, 2021-01-21:
支持替换，搞起来，这样客户端可以简化却又能灵活使用

WXfeng-boy, 2021-01-21:
支持替代

black-head, 2021-01-21:
支持替代

C222l, 2021-01-21:
支持替代！

swxk521, 2021-01-21:
支持，吧trojan，ss都替换了，精简精简

langliang3241, 2021-01-21:
支持替代

Myoko, 2021-01-21:
替换吧

xraylf, 2021-01-21:
支持替代

betaxab, 2021-01-21:
低端可以只有 xray。但 16M 不是应该废弃了吗？？？？？？？？？？

jiuqi9997, 2021-01-21:
我说，支持。强烈支持。

Fiber-Five, 2021-01-21:
替代

luojun853, 2021-01-21:
支持替代

gsblay, 2021-01-21:
支持替代 ！！

guanhemeng, 2021-01-21:
支持加入xray，那边的开发比较活跃能带来持续的新特性和维护

BOBINIUNIU, 2021-01-21:
我刚把trojan go 装好，你们又搞出这样一个新玩意。一定要问我支持不支持，我们怎么能不支持大佬？
既然大佬有钦定的意思，那就别按基本法了。16M的路由器？赶紧丢到垃圾桶里去吧，它们本来就应该在那里。

NagaseKouichi, 2021-01-21:
替代之后还能用Socks5套娃吗？

realsteam, 2021-01-21:
替换吧，一个能囊括的，就没必要搞多个。

haoweili, 2021-01-21:
替代吧

dolphinpaopao, 2021-01-21:
ss移动端目前的支持是比较完善易用的。私人环境，简洁方便的ss还是首选。希望还是能保留ss-libev或者切换到正在积极开发的rust版本，感谢。使用人数众多的机场一般才需要各种混淆套娃规避一定的运营风险。

bwtogo, 2021-01-21:
支持大佬

jxph1123, 2021-01-22:
强烈支持统一，节约空间。。

martin0607, 2021-01-22:
如果是SSR先用Sock5转发是不是就必须勾选Xray了呢？

helianling, 2021-01-22:
代替吧，xray就该一统天下。

helianling, 2021-01-22:
建议X86平台空间大，保持不变，其他平台可适度精简。

Marcotso, 2021-01-22:
支持替换

lanfenglin, 2021-01-22:
支持替换

SamoyeHN, 2021-01-22:
已经替代了，256M

lanfenglin, 2021-01-22:
支持替换

NeedforGit, 2021-01-22:
目前 Xray 1.22 的 trojan+xtls 玩游戏 貌似还无法 Fullcone, Trojan-go 的可以。所以可不可以晚些时候再替换？让Xray再更新几个版本。

lqhorochi, 2021-01-22:
既然全面功能,可以替代,越精简越好...喜欢简介方案

openwrt-try, 2021-01-22:
我说今天怎么编译都体积过大，唉，一声叹息

pop2006, 2021-01-22:
为什么要替换？不是编译的时候需要什么功能就自己加上去的吗？

badO1a5A90, 2021-01-23:
> 目前 Xray 1.22 的 trojan+xtls 玩游戏 貌似还无法 Fullcone, Trojan-go 的可以。所以可不可以晚些时候再替换？让Xray再更新几个版本。

从Xray v1.2.1 开始就支持fullcone了。详情可以参见
https://github.com/XTLS/Xray-core/releases/tag/v1.2.1

firkerword, 2021-01-23:
支持替代

Michaol, 2021-01-23:
支持Xray

Pastechn, 2021-01-23:
支持替换 充分考虑小rom

djmao, 2021-01-23:
大调加入eqos

djmao, 2021-01-23:
大雕加入eqos

y9858, 2021-01-23:
换 XRAY全能

[16 more replies]

closed 2021-01-25 by coolsnowwolf