coolsnowwolf/lede #6233: Flow Offloading 与virtio网卡问题

waterun, 2021-01-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
在使用virtio网卡时，如果打开Flow Offloading，端口映射到同一宿主机的另外一台虚拟机，如果内外网端口不一致的情况下，端口转发失败。内外网端口相同时，转发成功。这个问题排查了好几天，因为家里和单位环境基本一样，最后才发现单位的op打开了Flow Offloading。关闭FlowOffloading后，转发正常。

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
在使用virtio网卡时，如果打开Flow Offloading，端口映射到同一宿主机的另外一台虚拟机，如果内外网端口不一致的情况下，端口转发失败。内外网端口相同时，转发成功。这个问题排查了好几天，因为家里和单位环境基本一样，最后才发现单位的op打开了Flow Offloading。关闭FlowOffloading后，转发正常。

### (2) 路由器型号和固件版本

A：x86 PVE+ OpenWrt R20.3.19 + Windows Server2019

### (3) 详细日志

A：

hxlls, 2021-01-23:
反馈时候都不舍得全新安装次最新版本测试么？现在都是R21版本了，还反馈20年3月的版本问题。。。中途经历了无数次修复。。

coolsnowwolf, 2021-01-24:
这个已经修复了

4860575, 2021-01-26:
pve 的内核里貌似没有Flow Offloading模块吧？

waterun, 2021-02-02:
> 这个已经修复了

感谢大雕，好久没编译了，老版本配置太多了，不想折腾，多多包涵

closed 2021-06-20 by coolsnowwolf