coolsnowwolf/lede #6238: WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'xray', which does not exist

qqsir-dev, 2021-01-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'xray', which does not exist

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

feheadls, 2021-01-23:
编辑这个文件，package/feeds/helloworld/luci-app-ssr-plus/Makefile，把depend on后面的xray依赖那行删除即可。

Cathgao, 2021-01-24:
重新feeds update和install

SamoyeHN, 2021-01-24:
全删了，可以。

closed 2021-01-24 by coolsnowwolf

teapethu, 2021-01-24:
说移到feeds里了，但是好像feeds里也没有啊，是L大忘了更新feeds了吗？

dlj390, 2021-01-24:
same issue！

Cathgao, 2021-01-24:
去掉feeds里helloword的注释重新update和install就行了，多看看这边和那边的commit

dlj390, 2021-01-25:
@Cathgao feeds中的注释早就去掉了，update & install也反复做过了，这个问题依旧存在，不太明白你的建议。

dlj390, 2021-01-25:
解决了，要手动配置个软链接。

spypie, 2021-01-29:
> 解决了，要手动配置个软链接。

愿闻其详

fairok, 2021-11-05:
> 
> 
> 解决了，要手动配置个软链接。

这个大哥，怎么配置的啊？我也一样的错误