coolsnowwolf/lede #6243: "过渡包“initramfs-kernel.bin问题

gcphym, 2021-01-24:
编译好之后 刷入自己编译的”过渡包“ openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin 进不了后台，但是刷别人的就可以，然后再刷自己编译出的升级包也能正常进后台，为什么？

EnnawYang, 2021-01-24:
尽量使用官方的吧

AmadeusGhost, 2021-01-24:
可能是因为这个 initramfs-kernel.bin 太大了（

gcphym, 2021-01-24:
> 可能是因为这个 initramfs-kernel.bin 太大了（

好像是这个原因 我第一次编译的时候选了一大堆插件。结果编译出来的initramfs-kernel.bin30多m，第2次编译的时候只选了几个插件 编译出来20几m。第2次编译出来的可以进后台了 但是一上传升级包连接就断了。请问我该如何做才能编译出一个可用的过渡包？

gcphym, 2021-01-24:
> 尽量使用官方的吧

> 尽量使用官方的吧

在这里 https://downloads.openwrt.org/snapshots/targets/ramips/mt7621 找的initramfs-kernel.bin 刷进去也进不了后台 看路由器指示灯好像是卡住了 手动断电重启也没用

AmadeusGhost, 2021-01-24:
> 在这里 https://downloads.openwrt.org/snapshots/targets/ramips/mt7621 找的initramfs-kernel.bin 刷进去也进不了后台

snapshot 的固件默认不带 luci，也就是说默认只能 ssh 上去。

coolsnowwolf, 2021-01-24:
第一次的过渡包越小越好

closed 2021-01-25 by coolsnowwolf