coolsnowwolf/lede #6254: 系统日志中有uhttpd报错

roy63258015, 2021-01-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统日志内有uhttpd 报错

### (2) 路由器型号和固件版本

A：J1900软路由 
OpenWrt R21.1.18 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：on Jan 25 19:03:17 2021 daemon.info dnsmasq[7942]: Connected to system UBus
Mon Jan 25 19:03:21 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 25 19:03:21 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 25 19:03:24 2021 daemon.info dnsmasq[8386]: Connected to system UBus
Mon Jan 25 19:04:09 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 25 19:04:09 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 25 19:04:12 2021 daemon.info dnsmasq[8949]: Connected to system UBus
Mon Jan 25 19:04:16 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 25 19:04:16 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 25 19:04:19 2021 daemon.info dnsmasq[9391]: Connected to system UBus
Mon Jan 25 19:04:39 2021 daemon.err uhttpd[2808]: cat: can't open '/etc/bench.log': No such file or directory
Mon Jan 25 19:04:39 2021 daemon.err uhttpd[2808]: cat: can't open '/etc/bench.log': No such file or directory
Mon Jan 25 19:39:13 2021 daemon.err uhttpd[2808]: cat: can't open '/etc/bench.log': No such file or directory
Mon Jan 25 19:39:13 2021 daemon.err uhttpd[2808]: cat: can't open '/etc/bench.log': No such file or directory

Cathgao, 2021-01-26:
这个是跑分分数，不用管。强迫症可以进shell运行一次/etc/coremark.sh

roy63258015, 2021-01-26:
> 这个是跑分分数，不用管。强迫症可以进shell运行一次/etc/coremark.sh

好的，明白，谢谢

closed 2021-06-20 by coolsnowwolf