coolsnowwolf/lede #6268: 为x86 32位目标而编译的Naiveproxy失败

moralrebuild, 2021-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
在Ubuntu 18.04 LTS上为目标x86 32bit编译helloworld，如果不选naiveproxy可以编译成功，反之失败。
make menuconfig的目标选项如下：
  Target System (x86)  --->
  Subtarget (x86_64)  --->                                        
  Target Profile (Generic x86/64)  --->

### (1) 具体问题
Naiveproxy原作者的github上提供有可下载的最新版x86 32bit安装包，说明本软件支持x86 32bit 。 搜索编译错误信息，网上有人说是缺少gcc 32位库，证明似乎不是。

dpkg -l | grep gcc
```
ii  gcc                            4:7.4.0-1ubuntu2.3                  amd64        GNU C compiler
ii  gcc-7                          7.5.0-3ubuntu1~18.04                amd64        GNU C compiler
ii  gcc-7-base:amd64               7.5.0-3ubuntu1~18.04                amd64        GCC, the GNU Compiler Collection (base package)
ii  gcc-7-multilib                 7.5.0-3ubuntu1~18.04                amd64        GNU C compiler (multilib support)
ii  gcc-8-base:amd64               8.4.0-1ubuntu1~18.04                amd64        GCC, the GNU Compiler Collection (base package)
ii  gcc-multilib                   4:7.4.0-1ubuntu2.3                  amd64        GNU C compiler (multilib files)
ii  lib32gcc-7-dev                 7.5.0-3ubuntu1~18.04                amd64        GCC support library (32 bit development files)
ii  lib32gcc1                      1:8.4.0-1ubuntu1~18.04              amd64        GCC support library (32 bit Version)
ii  libgcc-7-dev:amd64             7.5.0-3ubuntu1~18.04                amd64        GCC support library (development files)
ii  libgcc1:amd64                  1:8.4.0-1ubuntu1~18.04              amd64        GCC support library
ii  libx32gcc-7-dev                7.5.0-3ubuntu1~18.04                amd64        GCC support library (x32 development files)
ii  libx32gcc1                     1:8.4.0-1ubuntu1~18.04              amd64        GCC support library (x32)
```

### (2) 路由器型号和固件版本

路由器：X86 Pentium 32位
固件版本：R20.12.12 最新

### (3) 详细日志

make -j1 V=s

```
[2089/2089] LINK ./naive
FAILED: naive
/home/fish/lede/staging_dir/hostpkg/bin/python3 "../../build/toolchain/gcc_link_wrapper.py" --output="./naive" -- ../../third_party/llvm-build/Release+Asserts/bin/clang++ -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=lld -Wl,--icf=all -Wl,--color-diagnostics -flto=thin -Wl,--thinlto-jobs=all -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy,cache_size=10\%:cache_size_bytes=10g:cache_size_files=100000 -Wl,-mllvm,-enable-dse-memoryssa=false -Wl,--lto-O0 -fwhole-program-vtables -m32 -Wl,-O2 -Wl,--gc-sections -rdynamic -nostdlib++ --sysroot=../../../../../../staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl -I/home/fish/lede/staging_dir/target-i386_pentium4_musl/usr/include -I/home/fish/lede/staging_dir/target-i386_pentium4_musl/usr/include/nss -I/home/fish/lede/staging_dir/target-i386_pentium4_musl/usr/include/nspr -L/home/fish/lede/staging_dir/target-i386_pentium4_musl/usr/lib -fsanitize=cfi-vcall -fsanitize=cfi-icall -pie -Wl,--disable-new-dtags -Wl,--dynamic-linker=/lib/ld-musl-i386.so.1 -L/home/fish/lede/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/home/fish/lede/staging_dir/target-i386_pentium4_musl/usr/lib --target=i386-openwrt-linux-musl -o "./naive" -Wl,--start-group @"./naive.rsp"  -Wl,--end-group  -ldl -lpthread -lrt -lnss3 -lnssutil3 -lsmime3 -lsoftokn3 -lplds4 -lplc4 -lnspr4 -lresolv -latomic
ld.lld: error: cannot open crtbeginS.o: No such file or directory
ld.lld: error: unable to find library -lgcc
ld.lld: error: unable to find library -lgcc
ld.lld: error: cannot open crtendS.o: No such file or directory
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
Makefile:141: recipe for target '/home/fish/lede/build_dir/target-i386_pentium4_musl/naiveproxy-88.0.4324.96-1/.built' failed
make[3]: *** [/home/fish/lede/build_dir/target-i386_pentium4_musl/naiveproxy-88.0.4324.96-1/.built] Error 1
make[3]: Leaving directory '/home/fish/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#3393.47#177.36#912.12
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/fish/lede'
package/Makefile:107: recipe for target '/home/fish/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile' failed
make[1]: *** [/home/fish/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fish/lede'
/home/fish/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
```

1715173329, 2021-01-28:
似乎是没有正确链接到编译器
看一下 staging_dir/target-i386_pentium4_musl/host/libltdl/bin/libtool 里面的 crtbeginS.o 文件指向何

moralrebuild, 2021-01-30:
@1715173329 
谢谢指教：
```
$ cat staging_dir/target-i386_pentium4_musl/host/libltdl/bin/libtool | grep crtbeginS.o
predep_objects="/home/fish/lede/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib/gcc/i486-openwrt-linux-musl/8.4.0/../../../../i486-openwrt-linux-musl/lib/crti.o /home/fish/lede/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib/gcc/i486-openwrt-linux-musl/8.4.0/crtbeginS.o"
```
另外查找缺少的文件：
```
 $ find . -type f -name crtbeginS.o
./staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/initial/lib/gcc/i486-openwrt-linux-musl/8.4.0/crtbeginS.o
./staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib/gcc/i486-openwrt-linux-musl/8.4.0/crtbeginS.o
./build_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/gcc-8.4.0-initial/gcc/crtbeginS.o
./build_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/gcc-8.4.0-initial/i486-openwrt-linux-musl/libgcc/crtbeginS.o
./build_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/gcc-8.4.0-final/gcc/crtbeginS.o
./build_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/gcc-8.4.0-final/i486-openwrt-linux-musl/libgcc/crtbeginS.o
```

1715173329, 2021-01-30:
因为 x86 平台各平台的链接器不尽相同，导致该问题
https://github.com/klzgrad/naiveproxy/commit/d64a92410d70747bd10313f1ece20ba161f1fdfb

很遗憾，Chromium 现在这堆都是硬编码进去的，无法自动适配
我有空了会研究一下通用解决方案

现在的一个解决办法是做一个 patch，将 `src/build/toolchain/linux/BUILD.gn` 里面的
`i386-openwrt-linux-musl` 更改为 `i486-openwrt-linux-musl`

moralrebuild, 2021-01-30:
@1715173329 
非常感谢您花时间提供的分析和帮助！尽管还没有满意解决方案，总算知道问题的方向了。如果有代码更新，请告诉我，我可以第一时间做测试。再次感谢！

1715173329, 2021-01-30:
> @1715173329
> 非常感谢您花时间提供的分析和帮助！尽管还没有满意解决方案，总算知道问题的方向了。如果有代码更新，请告诉我，我可以第一时间做测试。再次感谢！

您太客气了

closed 2021-02-20 by coolsnowwolf