coolsnowwolf/lede #6270: MWAN3 无法安装

gexqin, 2021-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：MWAN3 无法安装

### (2) 路由器型号和固件版本

A：R20.4.8-x64

### (3) 详细日志

A：Installing mwan3 (2.10.6-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/mwan3_2.10.6-1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mwan3:
 * 	kernel (= 5.4.92-1-435e8cacd26efb34941107c904ecfdde)
 * opkg_install_cmd: Cannot install package mwan3.

LeanFly, 2021-01-27:
看log是内核不对应

EnnawYang, 2021-01-27:
编译进去即可

gexqin, 2021-01-27:
我知道是内核版本不对，内核版本4.19，比openwrt官方的4.14还要新。我不知道编译进去是指的编译MWAN3还是指编译内核啊，作者大大这个定制版本定制了好多东西啊，牵一发动全身啊。

OldTaoge, 2021-01-28:
> 我知道是内核版本不对，内核版本4.19，比openwrt官方的4.14还要新。我不知道编译进去是指的编译MWAN3还是指编译内核啊，作者大大这个定制版本定制了好多东西啊，牵一发动全身啊。

编译时勾选mwan3，安装自己编译的版本

shunlongyang, 2021-03-08:
遇到了同样的问题，有大佬能帮忙共享一个mwan3 ipk么，感谢

closed 2021-06-20 by coolsnowwolf