coolsnowwolf/lede #6271: 今天偶然发现上网时间控制无法禁止微信

laowanwan, 2021-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
今天偶然发现上网时间控制无法禁止微信，其他应用包括qq都能禁止，微信不仅能登录还能传送文件，抓包后发现有tcp连接，端口50086，ping相关IP不通。查看iptables的配置也是正确的。
百思不得其解，望答复
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：OpenWrt R20.6.18 / LuCI Master (git-20.141.34051-f750b5f)

### (3) 详细日志

A：无

laowanwan, 2021-01-27:
经过测试，发现与广告屏蔽大师设置为全局模式有关，将模式修改为Plus+模式问题解决。
不过又发现，上网时间控制对走SSR的流量不起作用，我的SSR设置的是绕过国内IP模式

laowanwan, 2021-01-28:
编译并升级到最新固件，问题依旧。
经过无数次测试，发现一个可能有效的办法：在开启了上网时间控制之后，重启firewall，可能会立即生效，但不是100%成功。

LGA1150, 2021-01-29:
上网时间控制插入的 iptables 规则在 FORWARD 链上，开启透明代理后就不会经过 FORWARD 而是 INPUT OUTPUT

laowanwan, 2021-01-30:
> 上网时间控制插入的iptables规则在FORWARD链上，开启透明代理后就不会经过FORWARD或INPUT OUTPUT

但是我抓包发现微信流量的目标IP是国内的，按道理应该走iptables而且该被屏蔽。
实在不行只有不让这台电脑用透明代理了。
新的问题又来了，如何屏蔽成人网站？openwrt似乎只能用屏蔽域名的模式，但是url列表哪里找啊？

LGA1150, 2021-01-30:
> 在开启了上网时间控制之后，重启firewall，可能会立即生效

由此推断可能是没有拦截在控制时间段前建立的连接，运行下 `iptables-save -t filter` 看下规则

> 如何屏蔽成人网站

SSR DNS 用 1.1.1.3/1.0.0.3

laowanwan, 2021-01-30:
> iptables-save -t filter

看规则是有的，感觉确实是不能拦截在开启上网控制之前建立的连接，还是需要重启firewall生效。
此外上网控制确实无法屏蔽透明路由的流量，google可以上的。
1.1.1.3确实能屏蔽一些成人网站，有些则不行。谢谢啦。目前只有把windows/mac的控制打开，浏览器装插件看行不行了

closed 2021-06-20 by coolsnowwolf