coolsnowwolf/lede #6272: 最新编译的MT1300固件,无线网络用不了

rockey2020, 2021-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刷机进入后台管理后,手机搜索不到无线网,修改wifi名称之后,后台显示连接不了无线网

### (2) 路由器型号和固件版本

A：今天最新  mt1300

### (3) 详细日志

A：

rockey2020, 2021-01-27:
@dreamcmi

dreamcmi, 2021-01-28:
我也发现问题了，我是从openwrt/openwrt.git仓库直接搬过来的支持文件，还在解决ing

designer-fanfan, 2021-01-28:
> 了
我直接用mt1300编译成功，但是刷openwrt-ramips-mt7621-glinet_gl-mt1300-initramfs-kernel.bin文件可以刷进去，但软件包界面显示没有flash空间，7615e的驱动也编译进去了，硬件能识别，但wifi也不能用；刷openwrt-ramips-mt7621-glinet_gl-mt1300-squashfs-sysupgrade.bin无法开机，你这边有遇到类似问题么？求助

dreamcmi, 2021-01-28:
> > 了
> > 我直接用mt1300编译成功，但是刷openwrt-ramips-mt7621-glinet_gl-mt1300-initramfs-kernel.bin文件可以刷进去，但软件包界面显示没有flash空间，7615e的驱动也编译进去了，硬件能识别，但wifi也不能用；刷openwrt-ramips-mt7621-glinet_gl-mt1300-squashfs-sysupgrade.bin无法开机，你这边有遇到类似问题么？求助

是的，要通过uboot刷写要用kernel文件，网页升级可以用sysupgrade

dreamcmi, 2021-01-28:
这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....

designer-fanfan, 2021-01-28:
> 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....

试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？

dreamcmi, 2021-01-28:
> > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> 
> 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？

我试着可以

designer-fanfan, 2021-01-28:
> 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....

我之前联系了他们官方的工程师，第三方编译他们暂时没有技术支持

designer-fanfan, 2021-01-28:
> > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > 
> > 
> > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> 
> 我试着可以

所有硬件都正常么？有没有出现无线不能用，内存是0的问题？

dreamcmi, 2021-01-28:
> > > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > > 
> > > 
> > > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> > 
> > 
> > 我试着可以
> 
> 所有硬件都正常么？有没有出现无线不能用，内存是0的问题？

无线不能用是真事，我也不清楚具体原因

designer-fanfan, 2021-01-28:
> > > > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > > > 
> > > > 
> > > > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> > > 
> > > 
> > > 我试着可以
> > 
> > 
> > 所有硬件都正常么？有没有出现无线不能用，内存是0的问题？
> 
> 无线不能用是真事，我也不清楚具体原因

我之前刷了xwrt的固件，无线用起来没有问题。

dreamcmi, 2021-01-28:
> > > > > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > > > > 
> > > > > 
> > > > > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> > > > 
> > > > 
> > > > 我试着可以
> > > 
> > > 
> > > 所有硬件都正常么？有没有出现无线不能用，内存是0的问题？
> > 
> > 
> > 无线不能用是真事，我也不清楚具体原因
> 
> 我之前刷了xwrt的固件，无线用起来没有问题。

我去了解下xwrt的源码

designer-fanfan, 2021-01-28:
> > > > > > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > > > > > 
> > > > > > 
> > > > > > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> > > > > 
> > > > > 
> > > > > 我试着可以
> > > > 
> > > > 
> > > > 所有硬件都正常么？有没有出现无线不能用，内存是0的问题？
> > > 
> > > 
> > > 无线不能用是真事，我也不清楚具体原因
> > 
> > 
> > 我之前刷了xwrt的固件，无线用起来没有问题。
> 
> 我去了解下xwrt的源码

保持沟通交流，我是真小白，只能按照lean的源码教程来了，其他啥都不会

designer-fanfan, 2021-01-29:
> > > > > > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > > > > > 
> > > > > > 
> > > > > > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> > > > > 
> > > > > 
> > > > > 我试着可以
> > > > 
> > > > 
> > > > 所有硬件都正常么？有没有出现无线不能用，内存是0的问题？
> > > 
> > > 
> > > 无线不能用是真事，我也不清楚具体原因
> > 
> > 
> > 我之前刷了xwrt的固件，无线用起来没有问题。
> 
> 我去了解下xwrt的源码

请问，有进展么？

rockey2020, 2021-01-30:
> > > > > > > > 这个mt1300的支持特别迷，等我去问下官方的答复，我是openwrt新手....
> > > > > > > 
> > > > > > > 
> > > > > > > 试用uboot刷kernel？然后，在openwrt的页面升级sysupgrade？我之前试了失败了，你刷成功了么？
> > > > > > 
> > > > > > 
> > > > > > 我试着可以
> > > > > 
> > > > > 
> > > > > 所有硬件都正常么？有没有出现无线不能用，内存是0的问题？
> > > > 
> > > > 
> > > > 无线不能用是真事，我也不清楚具体原因
> > > 
> > > 
> > > 我之前刷了xwrt的固件，无线用起来没有问题。
> > 
> > 
> > 我去了解下xwrt的源码
> 
> 请问，有进展么？

先等等呗 大佬也要上班休息的

rockey2020, 2021-02-01:
@designer-fanfan @dreamcmi 无线问题是因为在编译的时候,没有勾选kmod-mt7615d kmod-mt7615d_dbdc两个模块,勾选就可以正常使用了

[image]

[image]

dreamcmi, 2021-02-01:
> @designer-fanfan @dreamcmi 无线问题是因为在编译的时候,没有勾选kmod-mt7615d kmod-mt7615d_dbdc两个模块,勾选就可以正常使用了
> 
> [image]
> 
> [image]

了解了，是mt1300的5g信号驱动那边没写进默认设置导致的。

dreamcmi, 2021-02-01:
已经提交pr 等大雕同意吧

rockey2020, 2021-02-02:
@dreamcmi 用了kmod-mt7615d kmod-mt7615d_dbdc之后  2.4g只有最高150  5g只有400  而且信号波动好大,ping的时候10-120之间来回跳动,路由器就在我旁边
[image]

dreamcmi, 2021-02-02:
可能官方有专门的patch吧，他们的固件也不开源，就很烦了

dreamcmi, 2021-02-03:
wifi修补已经同步到仓库

closed 2021-02-03 by rockey2020

JKSTAFF, 2021-11-10:
> wifi修补已经同步到仓库  
一个编译上的疑问，在包含了7615D和DBDC并同时使用mtwifi的luci时，wireless-tools会变成{M}选中状态，随即编译会产生如下错误：  
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency wireless-tools for mt_wifi
 * pkg_hash_fetch_best_installation_candidate: Packages for mt_wifi found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mt_wifi.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-mtwifi:
 *      wireless-tools
 * opkg_install_cmd: Cannot install package luci-app-mtwifi
```
此时若强制选中 wireless-tools 呈现为{*}选中标识，编译会产生如下错误：  
```
Collected errors:
 * check_conflicts_for: The following packages conflict with mt_wifi:
 * check_conflicts_for:         kmod-mt7615d_dbdc * 
 * opkg_install_cmd: Cannot install package mt_wifi.
 * check_conflicts_for: The following packages conflict with mt_wifi:
 * check_conflicts_for:         kmod-mt7615d_dbdc * 
 * opkg_install_cmd: Cannot install package luci-app-mtwifi.
```  

该错误在openwrt官方和lean仓库编译MT1300固件时都会发生。向请教一下编译7615DBDC闭源驱动和luci界面时正确选择依赖和配置的方式