coolsnowwolf/lede #6276: 关于libcrypto.so等问题

LoogCP, 2021-01-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时报libcrypto.so和libssl.so not found，但是系统里面有这两个库
### (2) 路由器型号和固件版本

A：X64软路由

### (3) 详细日志

A：
```
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: warning: libpthread.so.0, needed by /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libssl.so, not found (try using -rpath or -rpath-link)
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: warning: libc.so.6, needed by /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libssl.so, not found (try using -rpath or -rpath-link)
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: warning: libdl.so.2, needed by /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so, not found (try using -rpath or -rpath-link)
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `fdopen@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `mmap@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libssl.so: undefined reference to `strtol@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `sysconf@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `accept@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `__register_atfork@GLIBC_2.3.2'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libssl.so: undefined reference to `__stack_chk_fail@GLIBC_2.4'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `fopen64@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `close@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `stderr@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `__fread_chk@GLIBC_2.7'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `pthread_rwlock_rdlock@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `pthread_setspecific@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `pthread_key_delete@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `openlog@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `fputs@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `setsockopt@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `sigaction@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `getauxval@GLIBC_2.16'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `calloc@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `shmat@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libssl.so: undefined reference to `strchr@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `getenv@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `gethostbyname@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `__fdelt_chk@GLIBC_2.15'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `getsockname@GLIBC_2.2.5'
/home/lcp/Git/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/usr/lib/libcrypto.so: undefined reference to `pthread_rwlock_unlock@GLIBC_2.2.5'
[... cut ...]
make[3]: Leaving directory '/home/lcp/Git/lede/feeds/packages/net/smartdns'
time: package/feeds/packages/smartdns/compile#0.14#0.04#0.18
make[2]: *** [package/Makefile:113: package/feeds/packages/smartdns/compile] Error 2
make[2]: Leaving directory '/home/lcp/Git/lede'
make[1]: *** [package/Makefile:107: /home/lcp/Git/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lcp/Git/lede'
make: *** [/home/lcp/Git/lede/include/toplevel.mk:224：world] 错误 2
```

LGA1150, 2021-01-29:
是不是改过 libc 设置而没有 clean

LoogCP, 2021-01-29:
> 是不是改过 libc 设置而没有 clean

我没有改过libc，因为我也不知道怎么改

LoogCP, 2021-01-29:
> 是不是改过 libc 设置而没有 clean

我make clean试试

closed 2021-06-20 by coolsnowwolf