coolsnowwolf/lede #6281: SSR plus的PAC模式出现问题

jameswang510, 2021-01-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [×] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近几天上国内网站速度异常不稳定且非常慢，今天使用ip111.cn测试地址发现：访问国内网站使用的IP变成了外面IP地址，但是设置的一直都是PAC模式从未改变，更换不同节点后或重启软路由后重新测试会恢复正常（访问国内网址IP变为真实IP地址），但用不了多久再次测试所有IP地址又会变成外面的地址。这样的话PAC模式会和全局模式不就是没有区别了吗？

### (2) 路由器型号和固件版本

A：软路由购买于某宝（小马）
[image]

### (3) 详细日志

A：

ricksuzade-maker, 2021-01-28:
你这个版本这么老，都不会自己下载个最新版编译看看？

jameswang510, 2021-01-28:
> 你这个版本这么老，都不会自己下载个最新版编译看看？

是OPENWRT版本太老了吗？我这方面一知半解基本上不太懂的，但是想到之前一直稳定使用，想不明白问题出在哪里

usamr, 2021-02-01:
相同问题，在https://github.com/coolsnowwolf/lede/releases/tag/20200408
下载的 版本装在软路由虚拟机里面，GFW和绕过大陆模式开始正常，最近不好用了，什么都没设置，不知道如何解决

overprince, 2021-02-01:
是不是设置了自动更新gfwlist和CN ip list, 手动更新一次试试

jameswang510, 2021-02-01:
> 是不是设置了自动更新gfwlist和CN ip list, 手动更新一次试试

的确是设置了自动更新，但是我也手动更新过了，更新完之后短时间内会恢复正常，但是最后还是会变成全局模式的效果。现在升级openwrt后改用passwall了，暂时还没遇到问题

jameswang510, 2021-02-01:
> 相同问题，在https://github.com/coolsnowwolf/lede/releases/tag/20200408
> 下载的 版本装在软路由虚拟机里面，GFW和绕过大陆模式开始正常，最近不好用了，什么都没设置，不知道如何解决

我升级版本后SSR plus还是存在同样问题，现在改用passwall了

closed 2021-02-04 by coolsnowwolf