coolsnowwolf/lede #6289: 编译出错

luii2020, 2021-01-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译不成功

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：Configuring coreutils-nohup.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 *      xray
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-passwall:
 *      xray
 * opkg_install_cmd: Cannot install package luci-app-passwall.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:108: recipe for target '/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

kleisth, 2021-01-30:
同样的错误，把xray插件取消也一样报错

yonggedebaqi, 2021-01-30:
luci-app-passwall编译问题，你要去找luci-app-passwall作者。

luii2020, 2021-01-30:
我把xray的项目地址添加到feed后再编译就成功了，不明白git pull为什么不更新feed

ihuaer, 2021-01-30:
> 我把xray的项目地址添加到feed后再编译就成功了，不明白git pull为什么不更新feed

xray的项目地址能贴一下么

luii2020, 2021-01-30:
https://github.com/XTLS/Xray-core.git

EnnawYang, 2021-01-30:
**你们就不能update一下feeds吗？**

ihuaer, 2021-01-31:
> https://github.com/XTLS/Xray-core.git

谢谢，正在重新编译

cherrymars, 2021-01-31:
> **你们就不能update一下feeds吗？**

update feeds了也报这个错....

cherrymars, 2021-01-31:
> 我把xray的项目地址添加到feed后再编译就成功了，不明白git pull为什么不更新feeds

你好，你是加到feeds.conf文件还是怎么弄的？我加到feeds.conf或feeds.conf.default都报错

luii2020, 2021-01-31:
编辑feeds.conf.default，添加src-git xray https://github.com/XTLS/Xray-core.git，退出后更新feeds输入./scripts/feeds update -a
./scripts/feeds install -a

cherrymars, 2021-01-31:
> 编辑feeds.conf.default，添加src-git xray [https://github.com/XTLS/Xray-core.git，退出后更新feeds输入./scripts/feeds](https://github.com/XTLS/Xray-core.git%EF%BC%8C%E9%80%80%E5%87%BA%E5%90%8E%E6%9B%B4%E6%96%B0feeds%E8%BE%93%E5%85%A5./scripts/feeds) update -a
> ./scripts/feeds install -a

操作没错嘛，修改后更新feed直接报错，Unable to open feeds configuration at ./scripts/feeds line 91 见了个鬼

luii2020, 2021-01-31:
> 操作没错嘛，修改后更新feed直接报错，Unable to open feeds configuration at ./scripts/feeds line 91 见了个鬼

要不先试试执行rm -rf ./tmp && rm -rf .config删除临时文件和配置，再不行只能删掉 feeds 整个文件夹，在 lede 目录下执行 rm -rf ./feeds，然后再 update。

Cathgao, 2021-01-31:
`rm -rf ./feeds/&` 然后再update和install

ihuaer, 2021-02-01:
> `rm -rf ./feeds/&` 然后再update和install

感谢，有用啊

closed 2021-06-20 by coolsnowwolf