coolsnowwolf/lede #6293: 启动时如果插有非启动 U 盘，grub 会报错 disk "hd0.gpt1" not found

winkee01, 2021-01-30:
[image]

当把编译好的 img 文件刷入 U 盘等存储设备后，开机页面从该 U 盘启动(UEFI + GPT)，但是启动时如果电脑还插有其它非启动用的 U 盘，则 grub 自检时报错 disk "hd0.gpt1" not found，导致无法进入 openwrt 系统。

导致这个问题的原因是openwrt 中的 grub 造成的，因为 grub 会自检所有硬件，而发现某个硬件是非启动盘时，就报错这个设备，导致整个系统都无法启动。

目前解决办法是，在启动时把所有非启动用的 U 盘都拔掉，只留下用于启动 openwrt 的 U 盘。
长久的解决办法是，我个人认为是升级 openwrt 中的 grub，目前多数主流的 linux 发行版都用的是 grub2，所以我**猜测**如果升级成 grub2 可能会修复这个问题。

相关 ISSUE
#5196

EnnawYang, 2021-01-30:
顺便说下我的情况吧

nvme(openwrt + uefi) + sata3(ext4数据盘) 启动时也会存在这个问题，当把sata3盘移除后可正常开机，插回去后不能开机，所以我最后的解决办法是使用U盘当系统盘

sxx1314, 2021-01-30:
修改grub  使用uuid引导

winkee01, 2021-01-30:
经过昨晚一番调试，发现上述问题来自与 /boot/grub/grub.cfg 文件，其内容入下：
```
set default="0"
set timeout="0"
set root='(hd0,gpt1)'

menuentry "OpenWrt" {
        linux /boot/vmlinuz root=PARTUUID=684937a3-1ccc-0e03-30e8-37e24b0bfd02 rootwait   console=tty0 console=ttyS0,115200n8 noinitrd
}
```

报错提示来自于第三行的 set root='(hd0,gpt1)'，如果启动时插了多个外接磁盘（U盘），那么用于启动 openwrt 的 U 盘并不一定被识别为 hd0,gpt1，因此会报错。

在这行下面增加如下一行设置即可解决问题：
search --no-floppy --part-uuid --set=root <uuid>

注：
<uuid> 这里填写你启动分区的 uuid，启动分区的 uuid 可通过 blkid 命令查看，如果命令未找到，则执行 `opkg install blkid`。

liujixu1984, 2021-04-11:
这里不应该是吧第三行替换成这个内容吗？

closed 2021-04-15 by winkee01

billcxu, 2021-04-24:
> 这里不应该是吧第三行替换成这个内容吗？

我的是替换解决问题，直接新增没有作用，再次感谢@winkee01

passioki, 2021-07-21:
我也是替换解决了问题，感谢

megamindsam, 2021-12-27:
搞什么飞机，把第三行
set root='(hd0,gpt1)变成
#set root='(hd0,gpt1)，或者直接删除，保存退出，再重启就行了，看着显示屏亲测有效。我就是买的工控机自己装的内存条和一个120G的SATA当内存。
其他的改法，比如进入grub修改hd*，每次连插个键鼠的usb都会变，找系统盘的uuid我也不能确定是哪个。