coolsnowwolf/lede #6297: update_dnspod_cn.sh 不支持四级域名

shuosiw, 2021-01-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：配置了四级域名，比如 `xxx.yyy.zzz.com`使用 update_dnspod_cn.sh 更新会报错：

```
Error message:[域名不正确，请输入主域名，如 dnspod.cn]
```

经过验证，应该是取二级主域名时候异常，理论上应该是:

* `__DOMAIN=zzz.com`
* `__HOST=xxx.yyy`

也就是这行代码不兼容四级域名的 ddns：https://github.com/coolsnowwolf/lede/blob/master/package/lean/ddns-scripts_dnspod/update_dnspod_cn.sh#L17

```
[ "$domain" == "${domain/@/}" ] && domain="${domain/./@}" # 未找到分隔符，兼容常用域名格式
```

可以考虑修改为：

```
[ "$domain" == "${domain/@/}" ] && domain=`echo $domain | rev | sed 's/\./@/2'| rev`
```

替换倒数第二个点号，以兼容超过 三级域名的情况

### (2) 路由器型号和固件版本

A：k3

### (3) 详细日志

A：

```
 021522       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv6.dat -o /var/run/ddns/myddns_ipv6.err --no-check-certificate --no-proxy --post-data 'login_token=xxxxxxx,***PW***&format=json&domain=yyy.zzz.com&sub_domain=xxx' https://dnsapi.cn/Record.List
 021523 ERROR : Error message:[域名不正确，请输入主域名，如 dnspod.cn]
 021524  WARN : Transfer failed - 31 Try again in 60 seconds
```

shuosiw, 2021-01-30:
目前我修改了脚本，可以达到效果，但是实现方式可能比较挫：

```
[ "${domain:0:2}" == "@." ] && domain="${domain/./}" 
if [ "$domain" == "${domain/@/}" ]; then
    _dot_count=`echo $domain | grep -o '\.' | wc -l`                           # 判断域名有多少个点号
    domain=$(echo $domain | sed "s/\./@/`expr $_dot_count - 1`")  # 替换倒数第二个点号
fi
```

这几行代码执行完毕后的域名为：`xxx.yyy@zzz.com` 符合预期

同时，也兼容 3 级或 5 级域名：
```
# domain='xxx.yyy.zzz.com'; if [ "$domain" == "${domain/@/}" ]; then _dot_count=`echo $domain | grep -o '\.' | wc -l`; domain=$(echo $domain | sed "s/\./@/`expr $_dot_count - 1`");
fi; echo $domain
xxx.yyy@zzz.com
```
```
# domain='aaa.xxx.yyy.zzz.com'; if [ "$domain" == "${domain/@/}" ]; then _dot_count=`echo $domain | grep -o '\.' | wc -l`; domain=$(echo $domain | sed "s/\./@/`expr $_dot_count - 1`
"); fi; echo $domain
aaa.xxx.yyy@zzz.com
```

qcgzxw, 2021-01-31:
域名这样填就行了
xxx.yyy@zzz.com

shuosiw, 2021-02-06:
> 域名这样填就行了
> [xxx.yyy@zzz.com](mailto:xxx.yyy@zzz.com)

还有这种操作...

closed 2021-06-20 by coolsnowwolf