coolsnowwolf/lede #6299: 怎么开启路由器下所有端口都是通过https访问

yuanshus, 2021-01-31:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
需要https访问路由下的黑裙，下载机等设备

### (1) 具体问题
uhttpd只能访问路由管理后台是HTTPS，域名后面加端口就不行了

yuanshus, 2021-01-31:
已经有域名，证书

EnnawYang, 2021-01-31:
路由下的设备请另行部署证书

yuanshus, 2021-01-31:
> 路由下的设备请另行部署证书

路由器不能统一管理吗？

WYC-2020, 2021-01-31:
自己端口映射

yuanshus, 2021-01-31:
> 自己端口映射

可以说说详细操作吗？是不是要nginx反向代理？

dongjuanyong, 2021-01-31:
> > 路由下的设备请另行部署证书
> 
> 路由器不能统一管理吗？

可以，随便找个HTTP service muxer反代

yuanshus, 2021-01-31:
> > > 路由下的设备请另行部署证书
> > 
> > 
> > 路由器不能统一管理吗？
> 
> 可以，随便找个HTTP service muxer反代

我用uhttpd 只能访问路由管理界面，域名后面加端口就不行了。

dongjuanyong, 2021-01-31:
不行了指的是什么？http code是啥？防火墙端口开了没

yuanshus, 2021-01-31:
> 不行了指的是什么？http code是啥？防火墙端口开了没

直接无响应，不加端口只输入域名就https访问路由器管理页面

dongjuanyong, 2021-01-31:
你加的什么端口，这个端口在防火墙上怎么设置的

yuanshus, 2021-01-31:
> 你加的什么端口，这个端口在防火墙上怎么设置的

加的是http监听5000端口，防火墙端口转发5000到路由器下群晖的5000

dongjuanyong, 2021-01-31:
你都加转发了，路由上的监听就没用了（之监听lan了）
而且你转发之后，用不用ssl就只看群晖上的设置了

你需要做的，是路由上防火墙开放5000端口（不是转发），然后按个反代服务器，监听5000端口，启用ssl，反代到内网群晖

yuanshus, 2021-01-31:
> 你都加转发了，路由上的监听就没用了（之监听lan了）
> 而且你转发之后，用不用ssl就只看群晖上的设置了
> 
> 你需要做的，是路由上防火墙开放5000端口（不是转发），然后按个反代服务器，监听5000端口，启用ssl，反代到内网群晖

那我有个疑问，不转发只监听端口的话，我在外网访问域名：端口，路由怎么知道要访问内网机的那个IP：端口

dongjuanyong, 2021-01-31:
看反代的设置，以nginx为例：
```
server {
  listen       5000（监听端口） ssl;
  server_name  MY_SERVER_NAME（域名）;

  ssl_certificate /path/to/your/server/certificates/fullchain.pem;
  ssl_certificate_key /path/to/your/server/certificates/privkey.pem;
  location / {
    proxy_pass       http://192.168.1.88:5000（群晖ip加端口）;
  }
```

yuanshus, 2021-01-31:
> 看反代的设置，以nginx为例：
>
> ```
> server {
>   listen       5000（监听端口） ssl;
>   server_name  MY_SERVER_NAME（域名）;
>
>   ssl_certificate /path/to/your/server/certificates/fullchain.pem;
>   ssl_certificate_key [...]

原来如此，那我只要把需要走ssl的端口都取消端口转发，然后在nginx都设置监听端口和要代理的IP：端口，就可以解决我的问题了吗？

dongjuanyong, 2021-01-31:
是的，反代就是这么用的

WYC-2020, 2021-01-31:
到处都是保姆教程，有视频不看

yuanshus, 2021-01-31:
> 是的，反代就是这么用的
> 防火墙上别忘了打开端口

好的好的，谢谢分享

yuanshus, 2021-01-31:
> 到处都是保姆教程，有视频不看

可以分享个教程连接吗，我也是找了一天

yuanshus, 2021-01-31:
> > 是的，反代就是这么用的
> > 防火墙上别忘了打开端口
> 
> 好的好的，谢谢分享

yuanshus, 2021-01-31:
> 原来如此，那我只要把需要走ssl的端口都取消端口转发，然后在nginx都设置监听端口和要代理的IP：端口，就可以解决我的问题了吗？

openwrt好像默认是uhttpd，怎么改成nginx呢？

closed 2021-01-31 by yuanshus

dongjuanyong, 2021-01-31:
单装个nginx不管80和443，只监听你需要的其他端口当反代不就得了
或是装luci-ssl-nginx这个包用nginx代替uhttpd当luci的http server，最好提前拿虚拟机调调配置，很容易把整个luci弄挂了

yuanshus, 2021-01-31:
> 单装个nginx不管80和443，只监听你需要的其他端口当反代不就得了
> 或是装luci-ssl-nginx这个包用nginx代替uhttpd当luci的http server，最好提前拿虚拟机调调配置，很容易把整个luci弄挂了

我就是整挂了luci，打算重新编译。