coolsnowwolf/lede #6300: 为什么frp需要在整点和每个小时的第40分钟重启下

yunhai20082008, 2021-01-31:
计划任务有对应的项，但是不知道为什么要这么做
Sun Jan 31 08:00:00 2021 cron.err crond[6673]: USER root pid 24936 cmd /etc/init.d/frp restart
Sun Jan 31 08:00:00 2021 user.notice FRPC: Shutting down frp service
Sun Jan 31 08:00:00 2021 user.notice FRPC: Starting frp service
Sun Jan 31 08:40:00 2021 cron.err crond[6673]: USER root pid 30730 cmd /etc/init.d/frp restart
Sun Jan 31 08:40:00 2021 user.notice FRPC: Shutting down frp service
Sun Jan 31 08:40:00 2021 user.notice FRPC: Starting frp service
Sun Jan 31 09:00:00 2021 cron.err crond[6673]: USER root pid 1200 cmd /etc/init.d/frp restart
Sun Jan 31 09:00:00 2021 user.notice FRPC: Shutting down frp service
Sun Jan 31 09:00:00 2021 user.notice FRPC: Starting frp service
Sun Jan 31 09:40:00 2021 cron.err crond[6673]: USER root pid 7109 cmd /etc/init.d/frp restart
Sun Jan 31 09:40:00 2021 user.notice FRPC: Shutting down frp service
Sun Jan 31 09:40:00 2021 user.notice FRPC: Starting frp service

WYC-2020, 2021-01-31:
自己不知道看下那个心跳嘛关闭就是了

3703781, 2021-02-21:
试试"服务注册间隔"改为0，这个好像不是frp的功能，而是这个插件提供的

closed 2021-06-20 by coolsnowwolf

HellowBoy, 2025-02-07:
我是这样解决的，目前测试没问题了。
客户端和服务器端关闭端口复用，客户端注册间隔时间设置为0