coolsnowwolf/lede #6301: mkdir -p ./staging_dir/packages/ipq40xx  卡住

chenfuzhi, 2021-01-31:
反馈bug/问题模板，提建议请删除

###路由器型号和固件版本
**竞斗云2.0       P&W R619AC 128M**

### (3) 详细日志

make[5]: Entering directory '/root/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.92'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  Building modules, stage 2.
  MODPOST 0 modules
make[5]: Leaving directory '/root/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.92'
touch /root/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.92/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/root/lede-master/target/linux/ipq40xx/image'
make[5]: Nothing to be done for 'compile'.
make[5]: Leaving directory '/root/lede-master/target/linux/ipq40xx/image'
make[4]: Leaving directory '/root/lede-master/target/linux/ipq40xx'
make[3]: Leaving directory '/root/lede-master/target/linux'
time: target/linux/compile#0.80#0.14#5.25
make[2]: Leaving directory '/root/lede-master'
export MAKEFLAGS= ;make -w -r buildinfo
make[2]: Entering directory '/root/lede-master'
 make[3] diffconfig buildversion feedsversion
make[2]: Leaving directory '/root/lede-master'
make[2]: Entering directory '/root/lede-master'
rm -rf /root/lede-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx
make[2]: Leaving directory '/root/lede-master'
make[2]: Entering directory '/root/lede-master'
make[3]: Entering directory '/root/lede-master/package/libs/libjson-c'
make[3]: Leaving directory '/root/lede-master/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.03#0.00#0.15
make[3]: Entering directory '/root/lede-master/package/libs/libubox'
make[3]: Leaving directory '/root/lede-master/package/libs/libubox'
time: package/libs/libubox/host-compile#0.06#0.01#0.55
make[3]: Entering directory '/root/lede-master/package/system/opkg'
make[3]: Leaving directory '/root/lede-master/package/system/opkg'
time: package/system/opkg/host-compile#0.03#0.02#0.51
make[2]: Leaving directory '/root/lede-master'
make[2]: Entering directory '/root/lede-master'
rm -rf /root/lede-master/staging_dir/packages/ipq40xx
mkdir -p /root/lede-master/staging_dir/packages/ipq40xx

Aes64X, 2021-01-31:
有请下一位root神仙

chenfuzhi, 2021-01-31:
root不root，就是设置下 FORCE_UNSAFE_CONFIGURE=1  就好了。用正常用户也出现同样的问题

minlang112, 2021-01-31:
不要在root环境下编译……嗯

closed 2021-06-20 by coolsnowwolf