coolsnowwolf/lede #6305: SSR PLUS的trojan go没有办法生成配置文件导致无法允许

galaxyskyknight, 2021-01-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
见标题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无论用手工添加还是导入方式，软路由上ssrp里选择添加节点trojan-go，然后填入UUID,host,sni, 选用ws+tls,填好web路径之后，一切配置无误，点击添加应用，发现shadowsocksr起不来变红，观察/var/etc/ssrplus/tcp-udp-ssr-retcp.json 为空，没有生成配置，猜测哪里出错配置文件生成失败，不知道在哪里看日志。vps端配置肯定没问题，用Qv2ray同样的trojan-go协议可以连接。ssrp里该用同一个节点的vless没问题

### (2) 路由器型号和固件版本
Intel(R) Celeron(R) CPU 3865U @ 1.80GHz : 2 Core 2 Thread (CpuMark : 27583.115685 Scores)
OpenWrt R21.1.18 (2021-01-30) / LuCI Master (git-20.343.54716-6fc079f)
内核：5.4.92

A：

### (3) 详细日志
这里只能看到ssrp redir tcp error

2021-01-31 17:23:15: -----------end------------
2021-01-31 17:23:12: Main node:Trojan-Go 0.8.2 Started!
2021-01-31 17:23:11: ----------start------------
2021-01-31 17:23:11: ssrplus redir tcp error.restart!
2021-01-31 17:22:41: -----------end------------
2021-01-31 17:22:37: Main node:Trojan-Go 0.8.2 Started!
2021-01-31 17:22:37: ----------start------------
2021-01-31 17:22:36: ssrplus redir tcp error.restart!
2021-01-31 17:22:06: -----------end------------
2021-01-31 17:22:03: Main node:Trojan-Go 0.8.2 Started!
2021-01-31 17:22:02: ----------start------------
2021-01-31 17:22:02: ssrplus redir tcp error.restart!
2021-01-31 17:21:32: -----------end------------
2021-01-31 17:21:29: Main node:Trojan-Go 0.8.2 Started!
2021-01-31 17:21:28: ----------start------------
2021-01-31 17:21:28: ssrplus redir tcp error.restart!
2021-01-31 17:20:58: -----------end------------
2021-01-31 17:20:55: Main node:Trojan-Go 0.8.2 Started!
2021-01-31 17:20:54: ----------start------------
2021-01-31 17:20:53: ssrplus redir tcp error.restart!
2021-01-31 17:20:23: -----------end------------

A：

minlang112, 2021-01-31:
提错地方了，请到插件对应的项目页面提交。

jxph1123, 2021-02-04:
更改/etc/config/shadowsocksr的权限默认644为755后试试。
https://github.com/fw876/helloworld/issues/354

galaxyskyknight, 2021-02-04:
> 更改/etc/config/shadowsocksr的权限默认644为755后试试。

没用，我觉得应该是解析错误导致的

yutian33, 2021-03-03:
我也碰到同样的问题，求解 ~

closed 2021-06-20 by coolsnowwolf