coolsnowwolf/lede #6306: 编译错误

fishtorres, 2021-01-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译错误 python3相关？

### (2) 路由器型号和固件版本

A：最新源码 acrh-17

### (3) 详细日志

A：
Makefile:354: recipe for target '/home/fish/lede/build_dir/hostpkg/Python-3.8.3/.configured' failed
make[3]: *** [/home/fish/lede/build_dir/hostpkg/Python-3.8.3/.configured] Error 1
make[3]: Leaving directory '/home/fish/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#14.93#43.28#76.69
package/Makefile:111: recipe for target 'package/feeds/packages/python3/host/compile' failed
make[2]: *** [package/feeds/packages/python3/host/compile] Error 2
make[2]: Leaving directory '/home/fish/lede'
package/Makefile:107: recipe for target '/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fish/lede'
Build failed - please re-run with -j1 to see the real error message
/home/fish/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

jjit, 2021-01-31:
使用 make -j1 V=s 这个命令来编译的吗
错误信息也说是要用 -j1 参数来显示详细错误信息

fishtorres, 2021-02-01:
> 是用 make -j1 V=s 这个命令来编译的吗
> 错误信息也说是要用 -j1 参数来显示详细错误信息

不好意思，贴错了
make[2]: Leaving directory '/home/fish/lede'
make[2]: Entering directory '/home/fish/lede'
make[3]: Entering directory '/home/fish/lede/package/libs/libjson-c'
make[3]: Leaving directory '/home/fish/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.03#0.46#0.58
make[3]: Entering directory '/home/fish/lede/package/libs/libubox'
make[3]: Leaving directory '/home/fish/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#0.29#1.50#2.40
make[3]: Entering directory '/home/fish/lede/package/system/opkg'
CFLAGS="-O2 -I/home/fish/lede/staging_dir/host/include -I/home/fish/lede/staging_dir/hostpkg/include -I/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include" CPPFLAGS="-I/home/fish/lede/staging_dir/host/include -I/home/fish/lede/staging_dir/hostpkg/include -I/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include" CXXFLAGS="" LDFLAGS="-L/home/fish/lede/staging_dir/host/lib -L/home/fish/lede/staging_dir/hostpkg/lib -L/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/lib" make  -C /home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89
make[4]: Entering directory '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89'
make[5]: Entering directory '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89'
make[6]: Entering directory '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89'
libbb/CMakeFiles/bb.dir/flags.make:9: *** target pattern contains no '%'.  Stop.
make[6]: Leaving directory '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89'
CMakeFiles/Makefile2:152: recipe for target 'libbb/CMakeFiles/bb.dir/all' failed
make[5]: *** [libbb/CMakeFiles/bb.dir/all] Error 2
make[5]: Leaving directory '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89'
Makefile:146: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89'
Makefile:103: recipe for target '/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89/.built' failed
make[3]: *** [/home/fish/lede/build_dir/hostpkg/opkg-2020-05-07-f2166a89/.built] Error 2
make[3]: Leaving directory '/home/fish/lede/package/system/opkg'
time: package/system/opkg/host-compile#0.15#1.42#1.90
package/Makefile:111: recipe for target 'package/system/opkg/host/compile' failed
make[2]: *** [package/system/opkg/host/compile] Error 2
make[2]: Leaving directory '/home/fish/lede'
package/Makefile:107: recipe for target '/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/fish/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fish/lede'
/home/fish/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

chenfuzhi, 2021-02-01:
**libbb/CMakeFiles/bb.dir/flags.make:9: *** target pattern contains no '%'. Stop.**
貌似是没有all   编译选项

woalbb, 2021-02-02:
你这是不是也是硬件不足，内存不小于1G，硬盘剩余空间不小于10G再试试。

closed 2021-06-20 by coolsnowwolf