coolsnowwolf/lede #6309: 编译了一个早上，编译失败。

yuanshus, 2021-02-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
编译时候collections下的选项都选上了，包括：
luci
luci-nginx
luci-ssl-nginx
luci-ssl-openssl

因为我想用nginx做反向代理，所以选上的。
### (1) 具体问题

A：
Collected errors:
 * check_data_file_clashes: Package nginx-mod-luci-ssl wants to install file /home/yuanshu100/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/nginx/conf.d/luci.locations
        But that file is already provided by package  * nginx-mod-luci
 * check_data_file_clashes: Package nginx-mod-luci-ssl wants to install file /home/yuanshu100/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/uci-defaults/60_nginx-luci-support
        But that file is already provided by package  * nginx-mod-luci
 * opkg_install_cmd: Cannot install package luci-ssl-nginx.
 * check_data_file_clashes: Package nginx-mod-luci-ssl wants to install file /home/yuanshu100/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/nginx/conf.d/luci.locations
        But that file is already provided by package  * nginx-mod-luci
 * check_data_file_clashes: Package nginx-mod-luci-ssl wants to install file /home/yuanshu100/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/uci-defaults/60_nginx-luci-support
        But that file is already provided by package  * nginx-mod-luci
 * opkg_install_cmd: Cannot install package nginx-mod-luci-ssl.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/yuanshu100/lede'
package/Makefile:108: recipe for target '/home/yuanshu100/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/yuanshu100/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yuanshu100/lede'
/home/yuanshu100/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

XiaoliChan, 2021-02-01:
The source can't compile luci-ssl with luci [non-ssl] together.

yuanshus, 2021-02-01:
> The source can't compile luci-ssl with luci [non-ssl] together.
所以只能选择luci-ssl-nginx，luci-ssl-openssl，或者luci-nginx？

XiaoliChan, 2021-02-01:
Yep. Also luci-ssl-openssl maybe no compatible with luci-ssl-nginx.

yuanshus, 2021-02-01:
> Yep. Also luci-ssl-openssl maybe no compatible with luci-ssl-nginx.
luci-ssl，
luci [non-ssl]
这两个有区别吗？

XiaoliChan, 2021-02-01:
> > Yep. Also luci-ssl-openssl maybe no compatible with luci-ssl-nginx.
> > luci-ssl，
> > luci [non-ssl]
> > 这两个有区别吗？

Goto openwrt.org check some details.

yuanshus, 2021-02-01:
> > > Yep. Also luci-ssl-openssl maybe no compatible with luci-ssl-nginx.
> > > luci-ssl，
> > > luci [non-ssl]
> > > 这两个有区别吗？
> 
> Goto openwrt.org check some details.

只选择了luci和luci-nginx也依旧报错

XiaoliChan, 2021-02-01:
> 只选择了luci和luci-nginx也依旧报错
They are no compatible each other.

yuanshus, 2021-02-01:
> > 只选择了luci和luci-nginx也依旧报错
> 
> They are no compatible each other.
只能选择luci，luci-nginx或者（luci-ssl-nginx，luci-ssl-openssl）其中一个吗？

chenfuzhi, 2021-02-01:
选择两个重复的模块编译就会出现错误。删除一个

closed 2021-02-02 by yuanshus