coolsnowwolf/lede #6310: 想限制单个和总的容器对cpu的资源占用，防止导致主机太卡，应该怎么操作，docker-ce还有报错

yunhai20082008, 2021-02-01:
`Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER-USER':No such file or directory
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER-ISOLATION-STAGE-1':No such file or directory
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Mon Feb  1 08:07:00 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Command failed: Not found
Mon Feb  1 08:07:00 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Feb  1 08:07:02 2021 daemon.err uhttpd[3189]: cat: can't open '/etc/bench.log': No such file or directory
Mon Feb  1 08:07:02 2021 daemon.err uhttpd[3189]: cat: can't open '/etc/bench.log': No such file or directory
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.081696415Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.btrfs" error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
Mon Feb  1 08:07:04 2021 daemon.err modprobe: failed to find a module named aufs
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.085884168Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.aufs" error="modprobe aufs failed: "": exit status 255"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.086260528Z" level=warning msg="could not use snapshotter zfs in metadata plugin" error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter: skip plugin"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.086273780Z" level=warning msg="could not use snapshotter btrfs in metadata plugin" error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.086280003Z" level=warning msg="could not use snapshotter aufs in metadata plugin" error="modprobe aufs failed: "": exit status 255"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.111172581Z" level=warning msg="unable to modify root key limit, number of containers could be limited by this quota: open /proc/sys/kernel/keys/root_maxkeys: no such file or directory"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.122931221Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Mon Feb  1 08:07:04 2021 kern.err kernel: [   31.075797] overlayfs: filesystem on '/opt/docker/check-overlayfs-support024844204/upper' not supported as upperdir
Mon Feb  1 08:07:04 2021 daemon.err modprobe: failed to find a module named aufs
Mon Feb  1 08:07:04 2021 kern.err kernel: [   31.109940] overlayfs: filesystem on '/opt/docker/check-overlayfs-support075975963/upper' not supported as upperdir
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.156234660Z" level=error msg="AUFS was not found in /proc/filesystems" storage-driver=aufs
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.156956317Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.156974859Z" level=error msg="Failed to built-in GetDriver graph devicemapper /opt/docker"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.182445579Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.182547035Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Mon Feb  1 08:07:04 2021 daemon.err dockerd[3906]: time="2021-02-01T00:07:04.262355103Z" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"`

woalbb, 2021-02-02:
docker自带资源控制的，百度docker资源控制就知道了。可以对cpu、内存和io都能设置

NagaseKouichi, 2021-02-02:
portainer上不就可以设置吗

closed 2021-06-20 by coolsnowwolf