coolsnowwolf/lede #6314: 选择内核为4.14.195时的编译错误

dwj0, 2021-02-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
由于5.4的内核没有交换机选项，因此我在编译前将内核改为4.14来编译，我更改了target/linux/ramips/Makefile和target/linux/ramips/mt7621/target.mk.中的版本为4.14, 在编译中遇到以下问题请指教。

### (1) 具体问题

A：在 target/linux/ramips/mt7621/ 中没有config-4.14文件，我用config-5.4代替和把版本回退到4.14.180里面的4.14代替，编译时都需要手动选择一些信息，如
VPE loader support. (MIPS_VPE_LOADER) [N/y/?] (NEW) y
  Load VPE program into memory hidden from linux (MIPS_VPE_LOADER_TOM) [Y/n/?] (NEW) y
  Enable support for AP/SP API (RTLX) (MIPS_VPE_APSP_API) [N/y/?] (NEW) y
MIPS Coherent Processing System support (MIPS_CPS) [Y/n/?]
我需要怎么选择config-4.14,,或自动选择这些信息

B：编译内核时产生错误，提示build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h文件中kthread_create函数没有声明，我在这个文件中添加 #include <linux/kthread.h> 可顺利编译.
可以将这个改到原始文件中吗，或者请教一下这个文件的原始文件在哪

### (2) 路由器型号和固件版本

A：HC5962， MT7621， 4.14.195  git(9bc05c5ce)

### (3) 详细日志

A：问题B的日志
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h: In function 'mt76_worker_setup':
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h:73:12: error: implicit declaration of function 'kthread_create'; did you mean 'kobject_create'? [-Werror=implicit-function-declaration]
  w->task = kthread_create(__mt76_worker_fn, w, "mt76-%s %s",
            ^~~~~~~~~~~~~~
            kobject_create
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h:73:10: error: assignment to 'struct task_struct *' from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
  w->task = kthread_create(__mt76_worker_fn, w, "mt76-%s %s",
          ^
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h: In function 'mt76_worker_disable':
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h:102:2: error: implicit declaration of function 'kthread_park'; did you mean 'read_pnet'? [-Werror=implicit-function-declaration]
  kthread_park(w->task);
  ^~~~~~~~~~~~
  read_pnet
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h: In function 'mt76_worker_enable':
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h:111:2: error: implicit declaration of function 'kthread_unpark'; did you mean 'read_unlock'? [-Werror=implicit-function-declaration]
  kthread_unpark(w->task);
  ^~~~~~~~~~~~~~
  read_unlock
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h: In function 'mt76_worker_teardown':
/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/util.h:120:2: error: implicit declaration of function 'kthread_stop'; did you mean 'ether_setup'? [-Werror=implicit-function-declaration]
  kthread_stop(w->task);
  ^~~~~~~~~~~~
  ether_setup
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:327: /home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/mmio.o] Error 1
make[4]: *** [Makefile:1550: _module_/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6] Error 2
make[4]: Leaving directory '/home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.195'
make[3]: *** [Makefile:397: /home/xxxxx/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-12-18-4a90fdf6/.built] Error 2
make[3]: Leaving directory '/home/xxxxx/tmp/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#2.30#2.00#3.85
make[2]: *** [package/Makefile:113: package/kernel/mt76/compile] Error 2
make[2]: Leaving directory '/home/xxxxx/tmp/lede'
make[1]: *** [package/Makefile:107: /home/xxxxx/tmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xxxxx/tmp/lede'
make: *** [/home/xxxxx/tmp/lede/include/toplevel.mk:224: world] Error 2

dwj0, 2021-02-03:
新问题， 用4.14.180中的config-4.14编译4.14.195成功，但路由器无限重启。。。。。。

AmadeusGhost, 2021-02-03:
这里已经不支持 4.14 了，请移步 https://github.com/coolsnowwolf/openwrt

closed 2021-02-04 by coolsnowwolf