coolsnowwolf/lede #6323: 编译出错

lyrielau, 2021-02-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天编译出错

### (2) 路由器型号和固件版本

A：N1盒子 arm64/N1_Openwrt_R21.1.18_k5.4.93-flippy-52+o.img

### (3) 详细日志

A：Collected errors:
 * check_data_file_clashes: Package xray-core wants to install file /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/bin/v2ray
	But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package luci-app-passwall.
 * check_data_file_clashes: Package xray-core wants to install file /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/bin/v2ray
	But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package xray-core.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/simon/lede'
make[1]: *** [package/Makefile:108: /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/simon/lede'
make: *** [/home/simon/lede/include/toplevel.mk:224：world] 错误 2

ShoukakuChuuCC, 2021-02-04:
同

dlj390, 2021-02-04:
x86 同样错误

hue715, 2021-02-04:
v2ray和xray只能选一个

ShoukakuChuuCC, 2021-02-04:
> v2ray和xray只能选一个

试过了，无效，我只安装passwall，只勾选xray还是这样

hue715, 2021-02-04:
.config全文件搜索过了吗？
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray=n
CONFIG_PACKAGE_v2ray=n

smallprogram, 2021-02-04:
V2Ray与 xray-core冲突了

ihuaer, 2021-02-05:
注释掉v2ray 编译通过

xylz0928, 2021-02-05:
原因是v2ray和xray-core冲突，
把v2ray删除即可，参考下面脚本
v2ray的bin会由xray-core提供，.config里不用把v2ray取消选中

```
git clone https://github.com/xiaorouji/openwrt-passwall package/openwrt-passwall
rm -rf ./package/lean/dns2socks
rm -rf ./package/lean/ipt2socks
rm -rf ./package/lean/kcptun
rm -rf ./package/lean/microsocks
rm -rf ./package/lean/shadowsocksr-libev
rm -rf ./package/lean/simple-obfs
rm -rf ./package/lean/tcping
rm -rf ./package/lean/trojan
rm -rf ./package/lean/v2ray
rm -rf ./package/lean/v2ray-plugin
rm -rf ./package/openwrt-passwall/naiveproxy
rm -rf ./package/openwrt-passwall/tcping
rm -rf ./package/openwrt-passwall/xray-core
```

smallprogram, 2021-02-05:
#6327 坐等你这个PR Merged

smallprogram, 2021-02-06:
最终解决方案。
1. 去掉Network -> Porject V ->  V2ray
[image]
2. 添加Network -> xray-core - Xray-core Configuration ---> Provide V2ray Binary using XRay
[image]
3. 添加Network -> xray-geodata (可选可不选)

再编译就没毛病了。

closed 2021-02-06 by lyrielau