coolsnowwolf/lede #6330: zzz-default-settings中有关ssrplus无效 ?

axlexan, 2021-02-05:
前几天虚拟机玩崩溃了，于是重装

这几天commits有较多的更新，于是编译&更新且重置配置

不过发现之前有效的ssrplus相关uci命令均不生效，经测试ssh登录路由器执行zzz-default-settings文件中的相关命令是OK的.....

zzz-default-settings有 uci commit shadsocksr 这条命令哦，而且其他配置都是可以的（unupd，flowoffload等） ~

hue715, 2021-02-05:
由于源码更新，默认SSR-plus的config为空，/etc/uci-default/luci-ssr-plus运行后会生成SSR-plus的config，然而zzz-default-settings更名成99-default-settings，会先于luci-ssr-plus运行，导致99-default-settings中SSR-plus相关uci配置失败，请自行在zzz-default-settings头部增加“/etc/uci-default/luci-ssr-plus"
[image]

hue715, 2021-02-06:
> 前几天虚拟机玩崩溃了，于是重装
> 
> 这几天commits有较多的更新，于是编译&更新且重置配置
> 
> 发现之前有效的ssrplus相关uci命令均不生效，经测试ssh登录路由器执行zzz-default-settings文件中的相关命令是OK的.....
> 
> zzz-default-settings有 uci commit shadsocksr 这条命令哦，而且其他配置都是可以的（unupd，flowoffload等） ~
> 
> 

如果已解决，麻烦关闭这个issue吧

closed 2021-02-06 by axlexan