coolsnowwolf/lede #6333: ShadowsocksR Plus+ 未运行

ICLXL, 2021-02-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
 
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
* [ ] 类似issue [](https://github.com/coolsnowwolf/lede/issues/1342)

## 2. 详细叙述

### (1) 具体问题

A：ShadowsocksR Plus+ 未运行,根据issues/1342进行手动启动/etc/init.d/shadowsocksr start报错

### (2) 路由器型号和固件版本

A：Linux OpenWrt 5.4.92 #0 SMP Thu Feb 4 08:27:47 2021 mips GNU/Linux
B:  路由器型号Newifi-D2 CPU:MediaTek MT7621
C: 固件版本自行编译Master分支,最后一次commitId:fbdca3d67ef3ed02aff2227e9346bd565040ddd8
D:ShadowsocksR Plus因在lede,package下未找到,所以编译的是一个别人备份的github地址:[](https://github.com/SuLingGG/luci-app-ssr-plus.git)

### (3) 详细日志

A：iptables v1.8.4 (legacy): unknown option "--on-port"
Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.4 (legacy): unknown option "--on-port"
Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.4 (legacy): unknown option "--on-port"
Try `iptables -h' or 'iptables --help' for more information.

PlumWolf, 2021-02-05:
我有同样的问题

programmejava, 2021-02-05:
版本太老了，不匹配， lean的 ssr plus一直在代码里，git clone 代码之后到lede目录，修改feeds.conf.default这个文件，去掉helloworld那一行的#号，再 feeds update  -a feeds install -a就有了   以前的issues里有ssr 不见了的issue 。没事就经常看看issue和changelog

ICLXL, 2021-02-06:
> 版本太老了，不匹配， lean的 ssr plus一直在代码里，git clone 代码之后到lede目录，修改feeds.conf.default这个文件，去掉helloworld那一行的#号，再 feeds update -a feeds install -a就有了 以前的issues里有ssr 不见了的issue 。没事就经常看看issue和changelog

谢谢,解决了,但是编译出来的好像没有ssr-redir,现在依旧无法启动提示找不到 ssr-redir，无法启动.

jjit, 2021-02-06:
make menuconfig，Network，Web Servers/Proxies 里选中 shadowsocksr-libev-alt

ICLXL, 2021-02-06:
> make menuconfig，网络，Web服务器/代理里添加shadowsocksr-libev-alt

这个已添加,还是不行

programmejava, 2021-02-06:
如果你之前添加的旧版的ssr 并编译了，现在同步当前版本，就要删除之前的代码，并清除编译产生的中间文件，实在不行就重新拉代码。代码混乱总是出现迷一样的错误

ICLXL, 2021-02-06:
> 如果你之前添加的旧版的ssr 并编译了，现在同步当前版本，就要删除之前拉取ssr的代码，并清除编译产生的中间文件，实在不行就重新clone全新lede代码。代码混乱总是出现迷一样的错误

重新Reinstall 系统了
按照教程重新跑了一遍,还是不行
serverip:167.179.81.88
username:root
password:2Gr+ND}usTN5=$(u
大佬可以帮忙看下吗

programmejava, 2021-02-06:
好的，我准备登录上去看看  是不是用的这个型号Newifi-D2

programmejava, 2021-02-06:
> > 如果你之前添加的旧版的ssr 并编译了，现在同步当前版本，就要删除之前拉取ssr的代码，并清除编译产生的中间文件，实在不行就重新clone全新lede代码。代码混乱总是出现迷一样的错误
> 
> 重新Reinstall 系统了
> 按照教程重新跑了一遍,还是不行
> serverip:167.179.81.88
> username:root
> password:2Gr+ND}usTN5=$(u
> 大佬可以帮忙看下吗

我登录上去了，重新拉了代码 先了基本组件 ssr plus的组件全部选了， 用screen在后台编译，  你用screen -r lede恢复看进度
你现在改下系统密码

ICLXL, 2021-02-06:
好的

closed 2021-02-07 by ICLXL