coolsnowwolf/lede #6334: R2S 添加docker编译报错

vincheese, 2021-02-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：没加docker编译前能过，加了编译就报错

### (2) 路由器型号和固件版本

A：R2S 21.2.1

### (3) 详细日志

A：
/home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.92/scripts/dtc/dtc -O dtb -i/home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.92/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-rockpro64.dtb /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-rockpro64.dtb.tmp
rm -f /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-rockpro64.dtb.tmp
mkdir -p /home/vin/source/lede/bin/targets/rockchip/armv8 /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp
rm -rf /home/vin/source/lede/build_dir/target-aarch64_generic_musl/json_info_files
/home/vin/source/lede/staging_dir/host/bin/mksquashfs4 /home/vin/source/lede/build_dir/target-aarch64_generic_musl/root-rockchip /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs -nopad -noappend -root-owned -comp xz -Xpreset 9 -Xe -Xlc 0 -Xlp 2 -Xpb 2  -b 256k -p '/dev d 755 0 0' -p '/dev/console c 600 0 0 5 1' -processors 2
Parallel mksquashfs: Using 2 processors
Creating 4.0 filesystem on /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs, block size 262144.
Pseudo file "/dev" exists in source filesystem "/home/vin/source/lede/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
[=================================================================================================================|] 2792/2792 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 54406.81 Kbytes (53.13 Mbytes)
	25.74% of uncompressed filesystem size (211356.50 Kbytes)
Inode table size 20400 bytes (19.92 Kbytes)
	22.28% of uncompressed inode table size (91565 bytes)
Directory table size 25800 bytes (25.20 Kbytes)
	42.86% of uncompressed directory table size (60198 bytes)
Number of duplicate files found 311
Number of inodes 2614
Number of files 2061
Number of fragments 68
Number of symbolic links  409
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 143
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
/home/vin/source/lede/staging_dir/host/bin/make_ext4fs -L rootfs -l 167772160 -b 4096 -m 0 -J -T 1612427267 /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4 /home/vin/source/lede/build_dir/target-aarch64_generic_musl/root-rockchip/
Creating filesystem with parameters:
    Size: 167772160
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 5120
    Inode size: 256
    Journal blocks: 0
    Label: rootfs
    Blocks: 40960
    Block groups: 2
    Reserved blocks: 0
    Reserved block group size: 15
error: ext4_allocate_best_fit_partial: failed to allocate 5581 blocks, out of space?
make[5]: *** [/home/vin/source/lede/include/image.mk:356: /home/vin/source/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
make[5]: Leaving directory '/home/vin/source/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:26: install] Error 2
make[4]: Leaving directory '/home/vin/source/lede/target/linux/rockchip'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/vin/source/lede/target/linux'
time: target/linux/install#55.65#11.27#81.21
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/vin/source/lede'
make[1]: *** [target/Makefile:19: /home/vin/source/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/vin/source/lede'
make: *** [/home/vin/source/lede/include/toplevel.mk:224：world] 错误 2

NagaseKouichi, 2021-02-05:
out of space，应该是包的大小超过了，设置下分区大小应该就可以

vincheese, 2021-02-05:
> out of space，应该是包的大小超过了，设置下分区大小应该就可以

大佬 求详细  小白不是很懂

closed 2021-02-05 by vincheese