coolsnowwolf/lede #6338: SFE导致路由器频繁重启

AwesomeSky, 2021-02-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
（存在相同问题 issue已关闭，但BUG仍然存在（issue #5349 )
## 2. 详细叙述
X64软路由开启SFE时会不定时重启
### (1) 具体问题

A：开启SFE后 路由器会不定时在24小时内重启1-2次，关闭后已稳定四天运行，在issue #5349 下有他人相同情况测试，确认是SFE的问题

### (2) 路由器型号和固件版本

A：J1900 X64 OpenWrt R21.1.18 内核 5.4.92

### (3) 详细日志

A：重启后丢失

MGBB, 2021-02-07:
sfe在x86用处不大，建议不勾选

WYC-2020, 2021-02-07:
目前我也是最新源码编译，没遇到过这个情况，也是开启了sfe没自动重启过

NagaseKouichi, 2021-02-07:
> 目前我也是最新源码编译，没遇到过这个情况，也是开启了sfe没自动重启过

开了SFE？负载高的时候也不会重启吗

WYC-2020, 2021-02-07:
@NagaseKouichi 负载高？我负载没很高过，就常规用，平日用根本就不会达到很高。你是咋用的，跑了啥

jxph1123, 2021-02-08:
我用Offload，没重启过

github-xihe, 2021-02-08:
我也遇到过，开启sfe确实会有几小时重启一次的现象，关闭就不会。或者使用offload。

XiaoliChan, 2021-02-08:
pve平台没有这种事发生

NagaseKouichi, 2021-02-08:
> @NagaseKouichi 负载高？我负载没很高过，就常规用，平日用根本就不会达到很高。你是咋用的，跑了啥

主要挂了PT和跑了一些docker，比如京东签到的docker，在24点的时候跑的脚本比较多，openwrt开SFE的话会自动重启

NagaseKouichi, 2021-02-08:
> 我用Offload，没重启过

Offload不是给7621之类支持NAT硬件加速的设备用的吗？X86也能用？

XiaoliChan, 2021-02-08:
> Offload不是给7621之类支持NAT硬件加速的设备用的吗？X86也能用？

实际上，在不支持hwnat的设备上启用hwnat，使用的还是swnat

db-one, 2021-02-08:
我原来挂京东云路由器跑流量的时候也遇到过，随机重启的，基本不会跑超过一天，基本都是十来个小时然后突然重启了，但是我是删除了微信推送解决的，然后就再也没有重启过了，具体也不清楚是什么原因

liwenjie119, 2021-02-09:
我的x86设备也存在莫名其妙的重启问题。sfe是默认编译的吗？看来编译的时候需要取消掉

AwesomeSky, 2021-02-09:
> 我的x86设备也存在莫名其妙的重启问题。sfe是默认编译的吗？看来编译的时候需要取消掉

默认的应该是flow

NagaseKouichi, 2021-02-09:
> > 我的x86设备也存在莫名其妙的重启问题。sfe是默认编译的吗？看来编译的时候需要取消掉
> 
> 默认的应该是flow

X86架构的openwrt默认应该是SFE

AmadeusGhost, 2021-02-09:
> X86架构的openwrt默认应该是SFE

早就默认 flow 了

XiaoliChan, 2021-02-09:
> 早就默认 flow 了

好像在commit #https://github.com/coolsnowwolf/lede/commit/9bc05c5ce39c4dc34161af0e6940a622793b01e7 又选回sfe了

 [Line 28](https://github.com/coolsnowwolf/lede/blob/master/target/linux/x86/Makefile)

AmadeusGhost, 2021-02-09:
> 好像在commit #[9bc05c5](https://github.com/coolsnowwolf/lede/commit/9bc05c5ce39c4dc34161af0e6940a622793b01e7) 又选回sfe了

当我没说 (((

mjyhj, 2021-02-16:
同X86 SFE，N2600，也是不定时重启

jameyhhh, 2021-02-20:
同同X86 SFE，一天已不定时莫名重启3次了，之前一直用4.19内核版本的FLOW加速，近1年没有无故重启过。

xe5700, 2021-02-21:
我mt7621用SFE也出问题，因为FLOW加速映射端口有BUG。

AwesomeSky, 2021-03-09:
大雕现已开通反馈通道 issue #6489 ，此issue关闭 大家踊跃向大雕反馈

closed 2021-03-09 by AwesomeSky

L1943704135, 2021-08-30:
eSir固件，同x86 CPU 4415U 2G，开启SFE 平均10小时左右重启一次，关了之后已经稳定运行一周了