coolsnowwolf/lede #6341: lede阉割了802.11r？

ICLXL, 2021-02-07:
编译openwrt,Master分支最新的代码在设置无线的地方有启用802.11r,编译lede,Master分支最新的没有.

acooler15, 2021-02-07:
路由器无线是mt76的？

ICLXL, 2021-02-07:
路由器型号:newifiD2
cpu:MTK7621

AmadeusGhost, 2021-02-07:
你换开源的 mt76 驱动就有了，现在默认用的闭源驱动没有

ICLXL, 2021-02-07:
好的,谢谢

closed 2021-02-07 by ICLXL

acooler15, 2021-02-07:
> 路由器型号:newifiD2
> cpu:MTK7621

lean项目里默认使用的是闭源驱动，闭源驱动不支持802.11r。需要手动切换为开源驱动，编译时取消`Luci => Applications =>luci-app-mtwifi`、`Network => Wireless => mt_wifi`、`Kernel modules => Wireless Drivers => kmod-mt7603e`、`Kernel modules => Wireless Drivers => kmod-mt76x2e`，选择`Kernelmodules => Wireless Drivers => kmod-mt7603`、`Kernel modules => Wireless Drivers => kmod-mt76x2`、`Network => WirelessAPD => wpad`

ICLXL, 2021-02-07:
> lean项目里默认使用的是闭源驱动，闭源驱动不支持802.11r。需要手动切换为开源驱动，编译时取消`Luci => Applications =>luci-app-mtwifi`、`Network => Wireless => mt_wifi`、`Kernel modules => Wireless Drivers => kmod-mt7603e`、`Kernel modules => [...]

谢谢