coolsnowwolf/lede #6342: 最近把周边三个设备全升级，结果集体暴问题。

binbooly, 2021-02-07:
两个月前的老版本用着一直没问题，想着升级 Xray 1.2.4 就升级了家里还有公司的三个路由器，结果全部曝出问题。
1、极路由1S升级后，WiFi只能覆盖5米以内，超出5米后，要么就是连接不上，要么就是连接上后信号强度也还可以，但就是访问不了互联网。
2、极路由4升级后，用了大半天不到24小时突然就断网了，重启也不行，着急用网所以临时翻出箱底的老设备顶上去临时用了，还没来得及寻找突然断网的原因。
3、小米路由器3 pro升级后，电信300M带宽，以前跑speedtest.cn可以瞬间到达并稳定在350M以上，可升级后现在测试峰值就只有一次302M，大部分时间稳定在250M多，换了几个终端设备测试，也接上有线跑都一样。

Wilson-coder, 2021-02-07:
1和2不知道，3的话正常，最近几天我这边speedtest测速也不正常

binbooly, 2021-02-07:
问题1又进一步观察了一下，发现一个现象，我在系统里点击重启无效，会一直存在此问题。但直接断电再开机后会正常大概几个小时，然后等过几个小时后就又会出现 WIFI 超5米后严重掉速的问题。在没升级之前没有这个问题，也不确定是不是这次升级系统碰巧遇上硬件坏了。

问题2也进一步排查了一下，我把 IPv4 WAN 接口“关闭”（但好像没啥反应），之后又点击“连接”，后发现恢复正常了。

问题3目前也还没排查出什么猫腻，打10000说我主帐号线路确实是300M带宽，而我的另一条副线宽带才是200M。目前发现主帐号线路300M和副帐号线路200M的 speedtest.cn 测速基本一样，无线和有线测得的结果一样。200M带宽测速一直平稳在250M左右，300M带宽以前印象中有过370M左右的，而且能平稳在350M以上，而现在只能平稳在250M左右，有过两三次300M左右的峰值，但很快就下来了。

binbooly, 2021-02-07:
问题3等回头抽时间直接连电脑拨号再测试一下看。

cqtbg, 2021-02-08:
前两天的最新固件用在我的Y1S上，5G信号自己就没有了。

sky123456789sky, 2021-02-09:
极路由1S重刷后 openvpn不能用了 无线也不对 IPV6防火墙也出问题了.......

xiaocenter, 2021-03-16:
x86  系统重启按钮无效   断电重启才可以

closed 2021-06-17 by coolsnowwolf