coolsnowwolf/lede #6343: noddos web UI出现错误

qqsir-dev, 2021-02-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute template dispatcher target for entry '/admin/status/noddos'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'noddos/clients'.
A runtime error occured: [string "/usr/lib/lua/luci/view/noddos/clients.htm"]:16: bad argument #1 to 'input' (/var/lib/noddos/DeviceProfiles.json: No such file or directory)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

jjit, 2021-02-07:
noddos这个程序很久没有更新已经死了
官方：https://github.com/noddos/noddos/issues/55
noddos首次运行的时候，好像要连接 www.noddos.io
生成一个 DeviceProfiles.json 文件
现在 www.noddos.io 这个域名没有续费，被域名商回收了
也就生成不了 DeviceProfiles.json

qqsir-dev, 2021-02-07:
那就彻底从LUCI删除吧

closed 2021-02-20 by qqsir-dev