coolsnowwolf/lede #6344: 编译生成固件能不能自动附上日期

rshi0212, 2021-02-07:
建议编译生成固件自动附上日期而不被覆盖。

Suwipe, 2021-02-07:
自己修改default-settings，拉到最下面把echo "DISTRIB_REVISION='R21.2.1'" >> /etc/openwrt_release里面的R21.2.1改成当前日期，编译刷进系统后就能在主界面状态的固件版本看到了

rshi0212, 2021-02-07:
> 自己修改default-settings，拉到最下面把echo "DISTRIB_REVISION='R21.2.1'" >> /etc/openwrt_release里面的R21.2.1改成当前日期，编译刷进系统后就能在主界面状态的固件版本看到了

不是再页面版本看日期，是编译后生成固件文件的文件名附上日期。

lphgor, 2021-02-08:
https://github.com/coolsnowwolf/lede/blob/fbdca3d67ef3ed02aff2227e9346bd565040ddd8/include/image.mk#L40-L44

我觉得应该是改这里吧，添加一行：
```Makefile
IMG_PREFIX_DATE:=$(shell date +"%Y%d%m")
```
然后把`$(IMG_PREFIX_DATE)`添加到`IMG_PREFIX`的某一个位置。
没有经过测试，仅供参考。

github-xihe, 2021-02-08:
https://github.com/P3TERX/Actions-OpenWrt
P3TERX大佬的模板有，自己抄一下就好

closed 2021-06-17 by coolsnowwolf

wangweitung, 2022-01-15:
> https://github.com/coolsnowwolf/lede/blob/fbdca3d67ef3ed02aff2227e9346bd565040ddd8/include/image.mk#L40-L44
> 
> 我觉得应该是改这里吧，添加一行：
> 
> ```makefile
> IMG_PREFIX_DATE:=$(shell date +"%Y%m%d")
> ```
> 
> 然后把`$(IMG_PREFIX_DATE)`添加到`IMG_PREFIX`的某一个位置。 没有经过测试，仅供参考。

把$(shell date +"%Y%m%d")-加入40行等号以后。编译后得到的文件有年份和日期。
20220115-openwrt-x86-64-generic-squashfs-rootfs.img
~~~
IMG_PREFIX:=$(shell date +"%Y%m%d")-$(VERSION_DIST_SANITIZED)-$(IMG_PREFIX_VERNUM)$(IMG_PREFIX_VERCODE)$(IMG_PREFIX_EXTRA)$(BOARD)$(if $(SUBTARGET),-$(SUBTARGET))
~~~