coolsnowwolf/lede #6349: 编译时卡在v2-plugin

rilyuuj, 2021-02-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
编译到v2时产生如下报错

### (1) 具体问题

A：全部默认配置，只添加了world库进行编译，报错如下日志。尝试使用解决得issue方法依旧卡在这一步，确认网络环境无问题，可以通过网页访问

### (2) 路由器型号和固件版本

A：Allwinner A20/A3x/H3/R40
      Sinovoip Banana Pi M2 Ultra

### (3) 详细日志

A：

Building targets
go: finding module for package v2ray.com/core/features
go: finding module for package v2ray.com/core/common/signal/done
go: finding module for package v2ray.com/core/common/platform/filesystem
go: finding module for package v2ray.com/core/common/signal/semaphore
go: finding module for package v2ray.com/core/common/protocol/bittorrent
go: finding module for package v2ray.com/core/common/bitmask
go: finding module for package v2ray.com/core/main/confloader
go: finding module for package v2ray.com/core/common/protocol/udp
go: finding module for package v2ray.com/core/common/cmdarg
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/common/protocol/headers.go:6:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/bitmask
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/config.go:12:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/cmdarg
../../main.go:40:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/platform/filesystem
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/app/dispatcher/sniffer.go:7:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/protocol/bittorrent
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/transport/internet/udp/dispatcher.go:14:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/protocol/udp
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/common/log/logger.go:10:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/signal/done
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/common/log/logger.go:11:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/common/signal/semaphore
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/v2ray.go:12:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/features
/home/ubt/lede/dl/go-mod-cache/github.com/v2fly/v2ray-core@v4.34.0+incompatible/config.go:13:2: module v2ray.com/core@latest found (v4.15.0+incompatible, replaced by github.com/v2fly/v2ray-core@v4.34.0+incompatible), but does not contain package v2ray.com/core/main/confloader

Makefile:74: recipe for target '/home/ubt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/v2ray-plugin-1.8.0/.built' failed
make[3]: *** [/home/ubt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/v2ray-plugin-1.8.0/.built] Error 1
make[3]: Leaving directory '/home/ubt/lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#0.84#1.53#3.22
package/Makefile:111: recipe for target 'package/lean/v2ray-plugin/compile' failed
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/ubt/lede'
package/Makefile:107: recipe for target '/home/ubt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/ubt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubt/lede'
/home/ubt/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

XiaoliChan, 2021-02-08:
This log show you ”can not find module in url: v2ray.com“.
All the modules has already move to other website, try to give a feedback to project ["v2fly"](https://github.com/v2fly/v2ray-core).

**Why not use Xray instead of v2ray.**

rilyuuj, 2021-02-09:
> This log show you ”can not find module in url: v2ray.com“.
> All the modules has already move to other website, try to send a feedback to project ["v2fly"](https://github.com/v2fly/v2ray-core).
> 
> **But..... Why not use Xray instead of v2ray.**

您好，谢谢答复。这是编译"hello world" （勾选得是ssrp默认组件）产生得报错，所有feed都已经update和install了，但是还有这个报错，麻烦请再看一下。

XiaoliChan, 2021-02-09:
> 您好，谢谢答复。这是编译"hello world" （勾选得是ssrp默认组件）产生得报错，所有feed都已经update和install了，但是还有这个报错，麻烦请再看一下。

V2ray in upstream source code has been fully remove, please do this way to recomplie:

```
make distclean
git pull
./scripts/feeds update -a && ./scripts/feeds install -a
make menuconfig
```

rilyuuj, 2021-02-10:
> > 您好，谢谢答复。这是编译"hello world" （勾选得是ssrp默认组件）产生得报错，所有feed都已经update和install了，但是还有这个报错，麻烦请再看一下。
> 
> V2ray in upstream source code has been fully removed, please do this way to recompile:
> 
> ```
> make distclean     ##（Remove all the things: include your config）
> git pull
> ./scripts/feeds update -a && ./scripts/feeds install -a
> make menuconfig
> ```

您好，重新git pull使用默认config（只勾选xray）配置成功了，但是在LuCI-->Applications里，原来可以勾选得NeteaseMusic消失了，请问在Network -->NeteaseMusic[check UnblockNeteaseMusic.NodeJS, check UnblockNeteaseMusicGo] 勾选效果是一样得么？compliing with GOPROXY proxy 是否需要勾选。
形如aria2勾选了，是否还有在Network-->Download Manager再次勾选得呢？

XiaoliChan, 2021-02-10:
> 但是在LuCI-->Applications里，原来可以勾选得NeteaseMusic消失了，请问在Network -->NeteaseMusic[check UnblockNeteaseMusic.NodeJS, check UnblockNeteaseMusicGo] 勾选效果是一样得么？compliing with GOPROXY proxy 是否需要勾选。

Please do the feeds update again.

> 形如aria2勾选了，是否还有在Network-->Download Manager再次勾选得呢？

I did not use aria2 before.

fyfh, 2021-02-14:
取消勾选v2ray-plugin试试

rilyuuj, 2021-02-17:
> > 但是在LuCI-->Applications里，原来可以勾选得NeteaseMusic消失了，请问在Network -->NeteaseMusic[check UnblockNeteaseMusic.NodeJS, check UnblockNeteaseMusicGo] 勾选效果是一样得么？compliing with GOPROXY proxy 是否需要勾选。
> 
> Please do the feeds update again.
> 
> > 形如aria2勾选了，是否还有在Network-->Download Manager再次勾选得呢？
> 
> I did not use aria2 before.

确实是feed没有安装完整，重新来一次就好了

rilyuuj, 2021-02-17:
> 取消勾选v2ray-plugin试试

取消就能编译成功，但是重复编译得时候会影响UnblockNeteaseMusic得编译（同样是go编译失败错误信息），不知道什么原因。目前尝试取消compatible with goproxy.cn 再试一次。

HUiTHEME, 2021-02-18:
> 取消勾选v2ray-plugin试试

取消勾选v2ray-plugin，在ssr里，v2线路还可以用吗？

rilyuuj, 2021-03-16:
尝试了几次v2ray-plugin 都编译失败，应该是源的问题了。
现放弃使用，改用编译xray都正常。关闭issue

closed 2021-03-16 by rilyuuj