coolsnowwolf/lede #6352: UnblockNeteaseMusic  好像还劫持京东的地址

namebuhtig, 2021-02-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： UnblockNeteaseMusic  好像还劫持京东的地址

### (2) 路由器型号和固件版本

A：OpenWrt R21.2.1 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：打开UnblockNeteaseMusic，放一首歌，再登录jd.com就显示证书不对，点证书就显示是UnblockNeteaseMusic的证书。关掉这个插件就好了。
[image]

namebuhtig, 2021-02-08:
我好想明白了，我把京东签到插件删掉后就没问题了。UnblockNeteaseMusic用的nodejs 版本的话好像和京东签到有冲突

XiaoliChan, 2021-02-08:
Did u mean there are two different version of node have installed ?

lydasia, 2021-02-09:
以前发现这个还劫持知乎cdn，于是就关掉了。

namebuhtig, 2021-02-09:
感觉这两个插件都调用nodejs，出差错了

XiaoliChan, 2021-02-09:
> 感觉这两个插件都调用nodejs，出差错了

Well, the default binary of `UnblockNeteaseMusic` is the GOlang version.

I use it with `JD daily sign in`, it works ordinarily.

closed 2021-06-17 by coolsnowwolf