coolsnowwolf/lede #6353: 现在编译的固件状态概览页面看不到cpu的频率信息了么

xxp2277, 2021-02-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
无
## 2. 详细叙述

### (1) 具体问题

A：luci界面：状态，概览界面看不到cpu的频率信息

### (2) 路由器型号和固件版本

A：wndr3800

### (3) 详细日志

A：cat /proc/cpuinfo的信息中好像确实没有频率信息了，还能找回来么
system type             : Atheros AR7161 rev 2
machine                 : NETGEAR WNDR3700/WNDR3800/WNDRMAC
processor               : 0
cpu model               : MIPS 24Kc V7.4
BogoMIPS                : 531.66
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 16
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16
Options implemented     : tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit nan_legacy nan_2008 perf
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

Ljzkirito, 2021-02-09:
MIPS 一直都没有cpu频率信息吧，MIPS都是定频运行

xxp2277, 2021-02-09:
> 
> 
> MIPS 一直都没有cpu频率信息吧，MIPS都是定频运行

恩，但是原来这路由器默认用680MHz运行的，uboot超频到800MHz了，时间长了都快忘了，如果可能，还是希望能看到这个信息。

closed 2021-06-17 by coolsnowwolf