coolsnowwolf/lede #6354: 希望能加入IP限速插件比如eqos

djmao, 2021-02-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

LittleWuXinjiang, 2021-02-11:
EQOS，其它源里有提供，但貌似是不是源码太老了、不兼容新版OP了？我尝试编译过，开启后没有任何效果。我现在用SQM，感觉还不错。

supppig, 2021-02-12:
> EQOS，其它源里有提供，但貌似是不是源码太老了、不兼容新版OP了？我尝试编译过，开启后没有任何效果。我现在用SQM，感觉还不错。

关掉SFE Fast Path再试试

xiaobaishu1, 2021-02-12:
我用石像鬼QOS固件好用极了迅雷这流氓都能限制[image]

hyird, 2021-02-22:
> 我用石像鬼QOS固件好用极了迅雷这流氓都能限制[image]

这规则怎么写 看的一脸蒙

wlmqwyf, 2021-02-23:
> > EQOS，其它源里有提供，但貌似是不是源码太老了、不兼容新版OP了？我尝试编译过，开启后没有任何效果。我现在用SQM，感觉还不错。
> 
> 关掉SFE Fast Path再试试

 我用的是Flow Offload，开启或关闭都没作用。不知道是缺少组件还是怎么回事……

xiaobaishu1, 2021-02-23:
> > > EQOS，其它源里有提供，但貌似是不是源码太老了、不兼容新版OP了？我尝试编译过，开启后没有任何效果。我现在用SQM，感觉还不错。
> > 
> > 
> > 关掉SFE Fast Path再试试
> 
> 我用的是Flow Offload，开启或关闭都没作用。不知道是缺少组件还是怎么回事……

确实是这样 我就算什么插件都不选就只留一个SQM都一样没什么效果 只有石像鬼QOS有用 迅雷什么都能控制

gy39830, 2021-03-05:
> 我用石像鬼QOS固件好用极了迅雷这流氓都能限制[image]

求个石像鬼

wlmqwyf, 2021-03-08:
> > > > EQOS，其它源里有提供，但貌似是不是源码太老了、不兼容新版OP了？我尝试编译过，开启后没有任何效果。我现在用SQM，感觉还不错。
> > > 
> > > 
> > > 关掉SFE Fast Path再试试
> > 
> > 
> > 我用的是Flow Offload，开启或关闭都没作用。不知道是缺少组件还是怎么回事……
> 
> 确实是这样 我就算什么插件都不选就只留一个SQM都一样没什么效果 只有石像鬼QOS有用 迅雷什么都能控制

我这边用的SQM，有不错的效果了。

closed 2021-06-17 by coolsnowwolf