coolsnowwolf/lede #6355: luci-app-syncdial设置了多拨并上网但不能显示当前在线接口数量！

zxlhhyccc, 2021-02-09:
当前在线接口数量显示为0
[image]

star-o0O, 2021-02-09:
多拨插件已经很久没有维护了，等一个大佬，我可以提供多线多拨环境供测试。

zxlhhyccc, 2021-02-09:
> 多拨插件已经很久没有维护了，等一个大佬，我可以提供多线多拨环境供测试。

我是单线多拨环境，可以提供测试！

LittleWuXinjiang, 2021-02-11:
自己用macvlan创建多个wan口不就行了嘛……
ip link add link （网口） name （新建网口名） type macvlan
就可以了。

zxlhhyccc, 2021-02-11:
> 自己用macvlan创建多个wan口不就行了嘛……
> ip link add link （网口） name （新建网口名） type macvlan
> 就可以了。

能方便不方便？用插件就方便！！！

star-o0O, 2021-02-12:
> 自己用macvlan创建多个wan口不就行了嘛……
> ip link add link （网口） name （新建网口名） type macvlan
> 就可以了。

想用插件的主要原因不是建wan口，主要是并发拨号和掉线重拨，现在掉线重拨（断开并重拨所有wan）已经失效了。其他操作基本都是一次性的都好弄，但拨号就要不停重复操作才行了。

LittleWuXinjiang, 2021-02-12:
> > 自己用macvlan创建多个wan口不就行了嘛……
> > ip link add link （网口） name （新建网口名） type macvlan
> > 就可以了。
> 
> 想用插件的主要原因不是建wan口，主要是并发拨号和掉线重拨，现在掉线重拨（断开并重拨所有wan）已经失效了。其他操作基本都是一次性的都好弄，但拨号就要不停重复操作才行了。

哦，那感觉很麻烦呢！

shu0802, 2021-02-16:
> > 自己用macvlan创建多个wan口不就行了嘛……
> > ip link add link （网口） name （新建网口名） type macvlan
> > 就可以了。
> 
> 想用插件的主要原因不是建wan口，主要是并发拨号和掉线重拨，现在掉线重拨（断开并重拨所有wan）已经失效了。其他操作基本都是一次性的都好弄，但拨号就要不停重复操作才行了。

> > 自己用macvlan创建多个wan口不就行了嘛……
> > ip link add link （网口） name （新建网口名） type macvlan
> > 就可以了。
> 
> 想用插件的主要原因不是建wan口，主要是并发拨号和掉线重拨，现在掉线重拨（断开并重拨所有wan）已经失效了。其他操作基本都是一次性的都好弄，但拨号就要不停重复操作才行了。

> > 自己用macvlan创建多个wan口不就行了嘛……
> > ip link add link （网口） name （新建网口名） type macvlan
> > 就可以了。
> 
> 想用插件的主要原因不是建wan口，主要是并发拨号和掉线重拨，现在掉线重拨（断开并重拨所有wan）已经失效了。其他操作基本都是一次性的都好弄，但拨号就要不停重复操作才行了。

> > 自己用macvlan创建多个wan口不就行了嘛……
> > ip link add link （网口） name （新建网口名） type macvlan
> > 就可以了。
> 
> 想用插件的主要原因不是建wan口，主要是并发拨号和掉线重拨，现在掉线重拨（断开并重拨所有wan）已经失效了。其他操作基本都是一次性的都好弄，但拨号就要不停重复操作才行了。

电信双线5拨，一直都正常的使用。只是这个多线仅支持2条线的拨号。

closed 2021-06-17 by coolsnowwolf