coolsnowwolf/lede #6359: 为啥现在的luci里面没有frps和frpc了

blueboyggh, 2021-02-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：上周git pull拉回来的源码feeds update以后在luci里还有luci-app-frpc/s呢，怎么今天删了目录重新拉回来的源码feeds update以后就没有了？我在L大的luci仓库里看也没有frp了，是去掉了吗？

### (2) 路由器型号和固件版本

A：网件WNDR4300

sxx1314, 2021-02-11:
[frpc](https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-frpc) 
[frps](https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-frps) 

什么叫仓库里没有了，XD。眼神真好。

blueboyggh, 2021-02-11:
> [frpc](https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-frpc)
> [frps](https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-frps)
> 
> 什么叫仓库里没有了，XD。眼神真好。

咦我去？又有了...好吧

closed 2021-02-11 by blueboyggh