coolsnowwolf/lede #6363: MTK闭源驱动如何设置MAC白名单？

OROKIN, 2021-02-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：MTK闭源驱动如何设置 AP 的 MAC 白名单？是不是只有开源驱动可以设置？

### (2) 路由器型号和固件版本

A：Newifi D2

### (3) 详细日志

A：

acooler15, 2021-02-13:
设置AccessPolicy和AccessControlList，AccessPolicy的值有0（关闭）、1（白名单）、2（黑名单），AccessControlList的值的格式为`mac地址;mac地址;mac地址`,mac地址英文应小写

OROKIN, 2021-02-13:
> 设置AccessPolicy和AccessControlList，AccessPolicy的值有0（关闭）、1（白名单）、2（黑名单），AccessControlList的值的格式为`mac地址;mac地址;mac地址`,mac地址英文应小写

谢谢！请问 MAC 地址分隔符应该是 `:`（比如 `00:11:22:33:44:55`） 还是 `-`（比如 `00-11-22-33-44-55`）？

acooler15, 2021-02-13:
> > 设置AccessPolicy和AccessControlList，AccessPolicy的值有0（关闭）、1（白名单）、2（黑名单），AccessControlList的值的格式为`mac地址;mac地址;mac地址`,mac地址英文应小写
> 
> 谢谢！请问 MAC 地址分隔符应该是 `:`（比如 `00:11:22:33:44:55`） 还是 `-`（比如 `00-11-22-33-44-55`）？

[image]

acooler15, 2021-02-13:
[mtk_wi-fi_softap_software_programming_guide_v4.6](https://widora.io/_media/mtk_wi-fi_softap_software_programming_guide_v4.6.pdf) 可以看下文档

OROKIN, 2021-02-20:
> [mtk_wi-fi_softap_software_programming_guide_v4.6](https://widora.io/_media/mtk_wi-fi_softap_software_programming_guide_v4.6.pdf) 可以看下文档

谢谢~

closed 2021-02-20 by OROKIN