coolsnowwolf/lede #6365: newifi3d2如何关闭网口上的灯呢？

iseeyou, 2021-02-13:
网线接口的灯，不是led。
我之前用Padavan，可以关闭。

searchsyber, 2021-02-15:
作肾

iseeyou, 2021-02-15:
> 
> 
> 作肾

想要全黑的路由器 哈哈

leopardciaw, 2021-02-16:
网上有的，搜索下，具体的我不记得了。之前我弄过，SSH进去放一个文件、启动项设置开关。

acooler15, 2021-02-16:
可能需要在编译时修改dts文件，看了下pandavan，pandavan是通过其mtk_esw的命令工具实现的，但是我不了解mtk_esw如何工作的😢

iseeyou, 2021-02-16:
> 
> 
> 可能需要在编译时修改dts文件，看了下pandavan，pandavan是通过其mtk_esw的命令工具实现的，但是我不了解mtk_esw如何工作的😢

我尝试修改dts文件，但是好像不起作用。。。。

iseeyou, 2021-02-16:
> 
> 
> 网上有的，搜索下，具体的我不记得了。之前我弄过，SSH进去放一个文件、启动项设置开关。

帮忙回忆一下，我没有找到，我找到了一个说明：https://www.v2ex.com/t/383742 但是自己修改没起作用，可能是修改错文件了把，我修改的是  target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts

acooler15, 2021-02-17:
> > 网上有的，搜索下，具体的我不记得了。之前我弄过，SSH进去放一个文件、启动项设置开关。
> 
> 帮忙回忆一下，我没有找到，我找到了一个说明：https://www.v2ex.com/t/383742 但是自己修改没起作用，可能是修改错文件了把，我修改的是 target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts

mt7620与mt7621的不一样

iseeyou, 2021-02-17:
> 
> 
> > > 网上有的，搜索下，具体的我不记得了。之前我弄过，SSH进去放一个文件、启动项设置开关。
> > 
> > 
> > 帮忙回忆一下，我没有找到，我找到了一个说明：https://www.v2ex.com/t/383742 但是自己修改没起作用，可能是修改错文件了把，我修改的是 target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts
> 
> mt7620与mt7621的不一样

是的，不过肯定是可以关闭的，继续摸索~！

leopardciaw, 2021-02-18:
> > 网上有的，搜索下，具体的我不记得了。之前我弄过，SSH进去放一个文件、启动项设置开关。
> 
> 帮忙回忆一下，我没有找到，我找到了一个说明：https://www.v2ex.com/t/383742 但是自己修改没起作用，可能是修改错文件了把，我修改的是 target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts

找到地址了，不知道是否是你需要的：
https://blog.wangmao.me/openwrt-timing-off-led-lights.html
我之前按照这个操作成功了。

iseeyou, 2021-02-18:
> 
> 
> > > 网上有的，搜索下，具体的我不记得了。之前我弄过，SSH进去放一个文件、启动项设置开关。
> > 
> > 
> > 帮忙回忆一下，我没有找到，我找到了一个说明：https://www.v2ex.com/t/383742 但是自己修改没起作用，可能是修改错文件了把，我修改的是 target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts
> 
> 找到地址了，不知道是否是你需要的：
> https://blog.wangmao.me/openwrt-timing-off-led-lights.html
> 我之前按照这个操作成功了。

谢谢，这好像是led的，不是网口的。

LGA1150, 2021-02-18:
网口灯是交换机控制的，不受 7621 SoC GPIO 控制
想关掉的话得魔改交换机驱动，把交换机注册为 GPIO 控制器
https://github.com/LGA1150/openwrt/commit/32e5361b522b449cdc843869eea717f253f7f6dc

iseeyou, 2021-02-18:
> 
> 
> 网口灯是交换机控制的，不受 7621 SoC GPIO 控制
> 想关掉的话得魔改交换机驱动，把交换机注册为 GPIO 控制器
> [LGA1150/openwrt@32e5361](https://github.com/LGA1150/openwrt/commit/32e5361b522b449cdc843869eea717f253f7f6dc)

非常感谢 晚上试试

iseeyou, 2021-02-25:
> 
> 
> 网口灯是交换机控制的，不受 7621 SoC GPIO 控制
> 想关掉的话得魔改交换机驱动，把交换机注册为 GPIO 控制器
> [LGA1150/openwrt@32e5361](https://github.com/LGA1150/openwrt/commit/32e5361b522b449cdc843869eea717f253f7f6dc)

newifi3d2 如何改呢？我改了之后不想，LED配置也不见了。

LGA1150, 2021-02-25:
> newifi3d2 如何改呢？我改了不行，LED配置也不见了。

是不是只改了 dts 没打补丁？

iseeyou, 2021-02-25:
> 
> 
> > newifi3d2 如何改呢？我改了不行，LED配置也不见了。
> 
> 是不是只改了 dts 没打补丁？注意 5.4 内核 补丁应该放在 `target/linux/ramips/patches-5.4` 里面

你的源里面有我要的补丁吗？我没打过，现在不知道怎么创建补丁。
有教程吗？或者帮我生成一个补丁...

closed 2021-06-17 by coolsnowwolf