coolsnowwolf/lede #6368: vultr服务器编译错误，大佬们帮忙瞧瞧？

rainalong, 2021-02-13:
out of memory allocating 5611 bytes after a total of 790239176 bytes
Makefile:2345: recipe for target 's-automata' failed
make[5]: *** [s-automata] Error 1
make[5]: Leaving directory '/home/lede/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/gcc'
Makefile:4251: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/lede/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial'
Makefile:36: recipe for target '/home/lede/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/.built' failed
make[3]: *** [/home/lede/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Error 2
make[3]: Leaving directory '/home/lede/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#6.42#0.67#7.50
toolchain/Makefile:98: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/lede/lede'
toolchain/Makefile:96: recipe for target '/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/lede/lede'
/home/lede/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

rainalong, 2021-02-13:
服务器配置 cpu 1   ram 1G   ssd 25G

EnnawYang, 2021-02-14:
内存 硬盘都太小

xxwzan, 2021-02-14:
同错误

xxwzan, 2021-02-14:
make[4]: Entering directory '/home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1'
make[5]: Entering directory '/home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1'
make[6]: Entering directory '/home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1/libiberty'
make[6]: *** No rule to make target 'all'.  Stop.
make[6]: Leaving directory '/home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1/libiberty'
make[5]: *** [Makefile:7040: all-libiberty] Error 2
make[5]: Leaving directory '/home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1'
make[4]: *** [Makefile:850: all] Error 2
make[4]: Leaving directory '/home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1'
make[3]: *** [Makefile:69: /home/drem/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gdb-8.3.1/.built] Error 2
make[3]: Leaving directory '/home/drem/lede/toolchain/gdb'
time: toolchain/gdb/compile#0.10#0.00#0.11
make[2]: *** [toolchain/Makefile:100: toolchain/gdb/compile] Error 2
make[2]: Leaving directory '/home/drem/lede'
make[1]: *** [toolchain/Makefile:96: /home/drem/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/drem/lede'
make: *** [/home/drem/lede/include/toplevel.mk:224：world] 错误 2

minlang112, 2021-02-14:
out of memory  编译机起码需要2G内存、60G磁盘空间

rainalong, 2021-02-14:
> out of memory 编译机起码需要2G内存、60G磁盘空间

好的 谢谢大佬 我换下配置试试 感谢~

rainalong, 2021-02-14:
> 内存 硬盘都太小

好的 谢谢大佬  我换配置试试~谢谢~

rainalong, 2021-02-14:
换了配置后 编译成功 谢谢两位大佬。

closed 2021-02-14 by rainalong