coolsnowwolf/lede #6369: 网抑云解锁Go版本编译失败

EnnawYang, 2021-02-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如标题，目前发现有群友也有此问题

### (2) 路由器型号和固件版本

A：64位软路由，编译环境GitHub Actions ubuntu-18.04

### (3) 详细日志

A：
```
Copying files from /workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo into /workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.go_work/build/src/github.com/cnsilvan/UnblockNeteaseMusic
app.go
cache/cache.go
common/common.go
config/config.go
go.mod
go.sum
host/host.go
network/network.go
pool/pool.go
processor/crypto/crypto.go
processor/processor.go
provider/kugou/kugou.go
provider/kuwo/kuwo.go
provider/kuwo/kuwoDES.go
provider/migu/migu.go
provider/provider.go
proxy/proxy.go
utils/utils.go
version/version.go

Symlinking directories from /workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src into /workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.go_work/build/src
Error: /workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src/github.com/containerd/containerd is already a Go library
Error: /workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src/github.com/opencontainers/runc is already a Go library
......
Error: /workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src/github.com/docker/cli/service/logs is already a Go library
Error: /workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src/github.com/docker/libnetwork is already a Go library

CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/workdir/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/workdir/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/cnsilvan/UnblockNeteaseMusic" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/cnsilvan/UnblockNeteaseMusic" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo" GO_BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.go_work/build" GO_BUILD_BIN_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo=UnblockNeteaseMusicGo -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo=UnblockNeteaseMusicGo -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.go_work/build" GOCACHE="/workdir/openwrt/tmp/go-build" GOMODCACHE="/workdir/openwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash /workdir/openwrt/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1613006379' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X github.com/cnsilvan/UnblockNeteaseMusic/version.Version=0.2.7 -buildid '1613006379' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
go: downloading golang.org/x/text v0.3.3

Building targets
go: downloading golang.org/x/text v0.3.3
../../utils/utils.go:22:2: golang.org/x/text@v0.3.3: Get "https://proxy.golang.com.cn/golang.org/x/text/@v/v0.3.3.zip": stopped after 10 redirects

Makefile:77: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.built' failed
make[3]: Leaving directory '/workdir/openwrt/package/lean/UnblockNeteaseMusicGo'
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.built] Error 1
```
[image]

AmadeusGhost, 2021-02-14:
试试 `export GOPROXY=https://goproxy.cn` 再编译看看

Jayjjjjj, 2021-02-14:
v2ray_plugin 有同样问题 换阿里云代理正常

closed 2021-02-15 by coolsnowwolf

qd201211, 2021-02-15:
https://goproxy.cn尝试编译显示404 not found，更换为阿里的https://mirrors.aliyun.com/goproxy/编译通过

PR #6371 (2021-02-15): goproxy: replace unavailable goproxy.io
fix: #6369