coolsnowwolf/lede #6376: 今天编译一直失败

czxinshui, 2021-02-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：../../main.go:18:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: reading https://goproxy.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip: 404 Not Found
	server response: not found: github.com/v2fly/v2ray-core@v4.34.0+incompatible: invalid version: +incompatible suffix not allowed: module contains a go.mod file, so semantic import versioning is required

mcbbsandnz, 2021-02-15:
我也碰到这这个情况,,我去查了下这个网址,发现https://goproxy.cn/github.com网址404了

sxx1314, 2021-02-15:
Makefile里删除 goproxy 即可。goproxy很鸡肋，竟然能玩openwrt就没必要用这玩意了。

mcbbsandnz, 2021-02-15:
> Makefile里删除 goproxy 即可。goproxy很鸡肋，竟然能玩openwrt就没必要用这玩意了。
> [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)

我这边遇到的是v2ray-plugins编译失败,问题也是包的下载网页404了,我是手动删掉v2ray-plugins和依赖才编译成功的

sxx1314, 2021-02-15:
> > Makefile里删除 goproxy 即可。goproxy很鸡肋，竟然能玩openwrt就没必要用这玩意了。
> > [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)
> 
> 我这边遇到的是v2ray-plugins编译失败,问题也是包的下载网页404了,我是手动删掉v2ray-plugins和依赖才编译成功的

自己看报错  v2p 一样用的goproxy

EnnawYang, 2021-02-16:
https://github.com/coolsnowwolf/lede/issues/6373

czxinshui, 2021-02-16:
> Makefile文件里删除goproxy即可.goproxy很鸡肋，竟然能玩的OpenWrt就没必要用这玩意了。
> [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)
说一下完整目录地址吧。谢谢。

sxx1314, 2021-02-16:
> > Makefile文件里删除goproxy即可.goproxy很鸡肋，竟然能玩的OpenWrt就没必要用这玩意了。
> > [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)
> > 说一下完整目录地址吧。谢谢。

例子就在上面链接  谁报错就把goproxy改成n

czxinshui, 2021-02-16:
> > > Makefile文件里删除goproxy可以.goproxy很鸡肋，竟然能玩的OpenWrt就没必要用这玩意了。
> > > [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)
> > > 说一下完整目录地址吧。谢谢。
> 
> 例子就在上面链接谁报错就把goproxy改成n

我好像明白了。我试试

czxinshui, 2021-02-16:
> > > Makefile文件里删除goproxy可以.goproxy很鸡肋，竟然能玩的OpenWrt就没必要用这玩意了。
> > > [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)
> > > 说一下完整目录地址吧。谢谢。
> 
> 例子就在上面链接谁报错就把goproxy改成n

还是一样
Building targets
go: github.com/v2fly/v2ray-core@v4.34.0+incompatible: reading https://goproxy.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.mod: 404 Not Found
	server response: not found: github.com/v2fly/v2ray-core@v4.34.0+incompatible: invalid version: +incompatible suffix not allowed: module contains a go.mod file, so semantic import versioning is required

define Package/$(PKG_NAME)/config
config $(PKG_NAME)_INCLUDE_GOPROXY
	bool "Compiling with GOPROXY proxy"
	default n
这个已经改成N了。还是没用啊

souwei168, 2021-02-17:
最近发现一个问题，大雕的 PLUS+ ,和PASSWALL ，不能同时编译，不然就无法编译成功，就是最后封包时，报错，有没有解决办法？

closed 2021-02-17 by coolsnowwolf

sxx1314, 2021-02-17:
> > > > Makefile文件里删除goproxy可以.goproxy很鸡肋，竟然能玩的OpenWrt就没必要用这玩意了。
> > > > [关闭goproxy](https://github.com/coolsnowwolf/lede/commit/6b2e7c13fd8edf14bc5a7d1a1a91b50d8afdf468)
> > > > 说一下完整目录地址吧。谢谢。
> > 
> > 
> > 例子就在上面链接谁报错就把goproxy改成n
> 
> 还是一样
> Building targets
> go: github.com/v2fly/v2ray-core@v4.34.0+incompatible: reading https://goproxy.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.mod: 404 Not Found
> server response: not found: github.com/v2fly/v2ray-core@v4.34.0+incompatible: invalid version: +incompatible suffix not allowed: module contains a go.mod file, so semantic import versioning is required
> 
> 2-----------------------------------------------
> Fast-forward
> package/lean/UnblockNeteaseMusicGo/Makefile | 4 ++--
> package/lean/ddns-scripts_dnspod/update_dnspod_cn.sh | 4 ++--
> package/lean/v2ray-plugin/Makefile | 4 ++--
> target/linux/ramips/dts/mt7620a_youku_yk1.dts | 4 +++-
> target/linux/ramips/dts/mt7621_glinet_gl-mt1300.dts | 8 +++++---
> target/linux/ramips/dts/mt7621_ubnt_unifi-nanohd.dts | 3 ++-
> 6 files changed, 16 insertions(+), 11 deletions(-)
> 3-----------------------------------------
> 
> define Package/$(PKG_NAME)/config
> config $(PKG_NAME)_INCLUDE_GOPROXY
> bool "Compiling with GOPROXY proxy"
> default n
> 这个已经改成N了。还是没用啊

.config 需要goproxy

PR #6391 (2021-02-17): v2ray-plugin: fix build with goproxy
fix: #6376 , fix: #6382