coolsnowwolf/lede #6377: 用美国VPS编译失败

zzlinwq, 2021-02-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
编译出错
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
_STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_X64' '-DV8_EMBEDDER_STRING="-node.39"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DV8_USE_SNAPSHOT' '-DV8_CONCURRENT_MARKING' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_EMBEDDED_BUILTINS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' -I/home/vibit/lede/staging_dir/target-x86_64_musl/usr/include -I../deps/v8 -I../deps/v8/include -I/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/obj/gen/inspector-generated-output-root -I../deps/v8/third_party/inspector_protocol -I/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/obj/gen/torque-output-root -I/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/obj/gen/generate-bytecode-output-root  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/.deps//home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/api/api.o.d.raw   -c
virtual memory exhausted: Cannot allocate memory
tools/v8_gypfiles/v8_base_without_compiler.host.mk:599: recipe for target '/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/api/api.o' failed
make[5]: *** [/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/api/api.o] Error 1
rm 1a50ebc4e981002ad169b0eeea8cebcd9a191592.intermediate f0f78610d7bed739dce9a757c9389fff479725ea.intermediate e815294ad6236be3032b24bd0aed2b57d4e749db.intermediate c18cc13ef2a92dddd2f6163016a11bb5df92dc2c.intermediate
Makefile:104: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4'
Makefile:154: recipe for target '/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/.built' failed
make[3]: *** [/home/vibit/lede/build_dir/target-x86_64_musl/node-v12.18.4/.built] Error 2
make[3]: Leaving directory '/home/vibit/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#1611.38#184.71#4154.32
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/vibit/lede'
package/Makefile:107: recipe for target '/home/vibit/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/vibit/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vibit/lede'
/home/vibit/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

### (1) 具体问题
vultr家的VPS 
vibit@vultr:~/lede$ uname -a
Linux vultr.guest 4.15.0-132-generic #136-Ubuntu SMP Tue Jan 12 14:58:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 18 内存1G
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

WYC-2020, 2021-02-16:
还真是啥子鬼都提

github-xihe, 2021-02-16:
命令行输入 sudo apt-get update ，然后输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync

minlang112, 2021-02-16:
求求你不要在这么弱鸡的VPS上搞编译了，内存都不够……virtual memory exhausted: Cannot allocate memory

mcbbsandnz, 2021-02-16:
你这vps内存到底多小.....

Elletear, 2021-02-16:
#6368 换个🐔

zzlinwq, 2021-02-16:
> 求求你不要在这么弱鸡的VPS上搞编译了，内存都不够……virtual memory exhausted: Cannot allocate memory

内存1G

zzlinwq, 2021-02-16:
我换个配置看看

klever1988, 2021-02-16:
挂载个2g的swap就行了

zzlinwq, 2021-02-16:
换四核8G机器可以编译了。 请关闭这个issues

Elletear, 2021-02-17:
可以自己手动关的😥

closed 2021-02-17 by zzlinwq