coolsnowwolf/lede #6378: 需要一项功能，不知道有没有同需求的大佬是如何解决的

tikkacn, 2021-02-16:
是这样，本地对大多数121开头的ip段（非内网ip）QOS严重，科学上网非常卡，只有222开头的ip段速度比较流畅，有没有插件可以拨号到指定ip段，之前用过拨号脚本，效果不佳，主要就是wan口时常挂起，脚本容易中断。
当然如果有大佬有靠谱的脚本，也感激不尽！

zzlinwq, 2021-02-16:
你是心理作用吧。121和222开头的IP有区别？

db-one, 2021-02-16:
原来在koss的上面有这样的功能，可以手动指定需要拔到的IP段，如果不是就自动断开重播，只到拨到你需要的为止，你可以参考一下能不能移植过来，luci界面估计就不行了，koss用的是软件中心的，跟原版还不一样，你可以只拿到程序本身然后用脚本运行，或者直接用koss的固件

sxx1314, 2021-02-16:
自己写个shell 即可   只要你不怕等 慢慢拨即可

tikkacn, 2021-02-16:
> 你是心理作用吧。121和222开头的IP有区别？

有区别的
我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。

tikkacn, 2021-02-16:
> 原来在koss的上面有这样的功能，可以手动指定需要拔到的IP段，如果不是就自动断开重播，只到拨到你需要的为止，你可以参考一下能不能移植过来，luci界面估计就不行了，koss用的是软件中心的，跟原版还不一样，你可以只拿到程序本身然后用脚本运行，或者直接用koss的固件

之前就是从koolshare的lede转过来的，bug一堆，已经习惯了lean的固件，回头看下能不能提取出来。

hyird, 2021-02-16:
一个shell的事情 一个 while if的事情

zzlinwq, 2021-02-17:
> > 你是心理作用吧。121和222开头的IP有区别？
> 
> 有区别的
> 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。

aga加速aws是什么意思？

zzlinwq, 2021-02-17:
> > 你是心理作用吧。121和222开头的IP有区别？
> 
> 有区别的
> 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。

照理来说同一台bas出去路由应该一样的啊。和ip段没关系吧。

tikkacn, 2021-02-17:
> > > 你是心理作用吧。121和222开头的IP有区别？
> > 
> > 
> > 有区别的
> > 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。
> 
> 照理来说同一台bas出去路由应该一样的啊。和ip段没关系吧。

路由基本一致，但是就是这么奇怪，ping aga的话222开头的可以达到50ms以内，121开的基本超过200，应该是家宽被QOS了，222开头的好多政企宽带都在这个段上，基本没有被QOS，单位里固定ip的ip段也是222开，科学上网同样如丝顺滑。

zzlinwq, 2021-02-17:
> > > > 你是心理作用吧。121和222开头的IP有区别？
> > > 
> > > 
> > > 有区别的
> > > 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。
> > 
> > 
> > 照理来说同一台bas出去路由应该一样的啊。和ip段没关系吧。
> 
> 路由基本一致，但是就是这么奇怪，ping aga的话222开头的可以达到50ms以内，121开的基本超过200，应该是家宽被QOS了，222开头的好多政企宽带都在这个段上，基本没有被QOS，单位里固定ip的ip段也是222开，科学上网同样如丝顺滑。

aga是什么东东？

tikkacn, 2021-02-17:
> > > > > 你是心理作用吧。121和222开头的IP有区别？
> > > > 
> > > > 
> > > > 有区别的
> > > > 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。
> > > 
> > > 
> > > 照理来说同一台bas出去路由应该一样的啊。和ip段没关系吧。
> > 
> > 
> > 路由基本一致，但是就是这么奇怪，ping aga的话222开头的可以达到50ms以内，121开的基本超过200，应该是家宽被QOS了，222开头的好多政企宽带都在这个段上，基本没有被QOS，单位里固定ip的ip段也是222开，科学上网同样如丝顺滑。
> 
> aga是什么东东？

aws global accelerator

z826540272, 2021-02-17:
> > 你是心理作用吧。121和222开头的IP有区别？
> 
> 有区别的
> 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。

我用的也是AGA，我自己的ip拨号上去，速度大概15000kbps，省会电信有50000+，其他省的有直接跑到10万的

z826540272, 2021-02-17:
> > > > > > 你是心理作用吧。121和222开头的IP有区别？
> > > > > 
> > > > > 
> > > > > 有区别的
> > > > > 我用aga加速aws，121开头的油管大概只能跑到222开头的1/50的速度，就是这么夸张。
> > > > 
> > > > 
> > > > 照理来说同一台bas出去路由应该一样的啊。和ip段没关系吧。
> > > 
> > > 
> > > 路由基本一致，但是就是这么奇怪，ping aga的话222开头的可以达到50ms以内，121开的基本超过200，应该是家宽被QOS了，222开头的好多政企宽带都在这个段上，基本没有被QOS，单位里固定ip的ip段也是222开，科学上网同样如丝顺滑。
> > 
> > 
> > aga是什么东东？
> 
> aws global accelerator

超过200肯定是有问题的，看路由记录

z826540272, 2021-02-17:
我建议你直接买个中转机场。

tikkacn, 2021-02-18:
> 我建议你直接买个中转机场。

中转机场还是算了，我这边几乎所有的都用过，效果最佳的就是这个aga，反正白嫖的效果又好，油管能跑到30w，只是从上个月开始就挑ip段了。

closed 2021-06-17 by coolsnowwolf