coolsnowwolf/lede #6382: 编译报错

HUiTHEME, 2021-02-16:
go: github.com/v2fly/v2ray-core@v4.34.0+incompatible: reading https://goproxy.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.mod: 404 Not Found
	server response: not found: github.com/v2fly/v2ray-core@v4.34.0+incompatible: invalid version: +incompatible suffix not allowed: module contains a go.mod file, so semantic import versioning is required

Building targets
go: github.com/v2fly/v2ray-core@v4.34.0+incompatible: reading https://goproxy.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.mod: 404 Not Found
	server response: not found: github.com/v2fly/v2ray-core@v4.34.0+incompatible: invalid version: +incompatible suffix not allowed: module contains a go.mod file, so semantic import versioning is required

make[3]: *** [Makefile:77: /home/zooo/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.8.0/.built] Error 1
make[3]: Leaving directory '/home/zooo/lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#0.23#0.16#0.70
make[2]: *** [package/Makefile:113: package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/zooo/lede'
make[1]: *** [package/Makefile:107: /home/zooo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zooo/lede'
make: *** [/home/zooo/lede/include/toplevel.mk:224: world] Error 2

camino13, 2021-02-17:
network->v2ray-plugin里面取消勾选goproxy

doosit, 2021-02-17:
> network->project v->v2ray-plugin里面取消勾选goproxy

感谢 重新拉取中

HUiTHEME, 2021-02-17:
> network->project v->v2ray-plugin里面取消勾选goproxy

我尝试了，在 LuCI -> Applications -> luci-app-ssr-plus 里，啥也不选，就默认状态，即（xray勾选，v2ray不选）的情况；编译是成功的；
按你说的 在  > network->project v->  把 compiling with GOPROXY proxy 取消勾选后，再在LuCI -> Applications ->  luci-app-ssr-plus 里，就可以把v2ray勾选了吗？

还有什么计划是什么？准备废弃v2ray？以后都采用xray？还是？？我看好多人出现这个情况，但基本被无视了。

ftfkdz66, 2021-02-17:
> 我尝试了，在 LuCI -> Applications -> luci-app-ssr-plus 里，啥也不选，就默认状态，即（xray勾选，v2ray不选）的情况；编译是成功的；
> 按你说的 在 > network->project v-> 把 compiling with GOPROXY proxy 取消勾选后，再在LuCI -> Applications -> [...]

取消勾选goproxy后，ssr-plus里面就能勾选v2ray-plugin

HUiTHEME, 2021-02-17:
测试结果：
一、 LuCI -> Applications -> luci-app-ssr-plus 的默认状态，xray勾选，v2ray不选的情况下，编译固件顺利，且使用过程中v2ray线路可以使用；
二、按照上述说的 Network -> project v -> 取消勾选goproxy，编译固件也顺利；且使用过程中v2ray线路也正常使用。

czxinshui, 2021-02-17:
> 测试结果：
> 一、 LuCI -> Applications -> luci-app-ssr-plus 的默认状态，xray勾选，v2ray不选的情况下，编译固件顺利，且使用过程中v2ray线路可以使用；
> 二、按照上述说的 Network -> project v -> 取消勾选goproxy，编译固件也顺利；且使用过程中v2ray线路也正常使用。

编译的时候取消Include Shadowsocks V2ray Plugin  的选择 才能成功，选中再用你说的第2种方法，也不行

closed 2021-02-17 by coolsnowwolf

HUiTHEME, 2021-02-18:
虽然 该 Issues 被关闭了，但结论还是没有得出；
Issues被关闭后，我今早又重新git pull，也看到源码有对v2相关的更新，于是开始重新编译了，今早的做法是把 LuCI -> Applications -> luci-app-ssr-plus 里的几个都勾选了(这也是我以往的操作)，但依然报错；
暂时的结论是， luci-app-ssr-plus 里的 V2ray Plugin不能勾选了？只能用Xray了吗？

PR #6391 (2021-02-17): v2ray-plugin: fix build with goproxy
fix: #6376 , fix: #6382