coolsnowwolf/lede #6384: luci-app-arpbind arp绑定不起作用，要修改 /etc/init.d/arpbind 33行才行。大家都OK？

iseeyou, 2021-02-16:
大家的luci-app-arpbind能用吗？我的不起作用，修改 /etc/init.d/arpbind 第33行 REACHABLE 才会变为 permanent。
把33行换成
ip neigh add $1 lladdr $2 nud permanent dev $3 || ip neigh change $1 lladdr $2 nud permanent dev $3
才起作用...

huah0235, 2021-02-18:
没具体测试，不过在SBR AC1750（QCA9558）上好像是有那么一次，ipmac绑定设置了，重启后IP依然在变化

iseeyou, 2021-02-18:
> 没具体测试，不过在SBR AC1750（QCA9558）上好像是有那么一次，ipmac绑定设置了，重启后IP依然在变化

有问题 要加个判断是不是已经有记录了，没有就add 有了就  change
不然如果dhcp分配了有记录还是add就不行

msylgj, 2021-02-22:
测试发现luci界面编辑配置之后不会直接生效, 手动start一下init.d/arpbind就好了, 修改了一下luci-app-arpbind/luasrc/model/cbi/arpbind.lua在return之前增加以下行,详细参见https://github.com/msylgj/OpenWrt_luci-app/commit/97e619d5c4fe5260e7f4ffad0bada383f0679d42

```
local e=luci.http.formvalue("cbi.apply")
if e then
  io.popen("/etc/init.d/arpbind start")
end
```

closed 2021-02-23 by iseeyou

msylgj, 2021-02-23:
重新测试了一下, 发现直接用脚本start时还是用旧配置生效的, 因此重写了cbi.apply时的处理逻辑, 提交时马上生效. 详见https://github.com/msylgj/OpenWrt_luci-app/commit/061b9efbdb959a7bf25c0779edd1158c611771ff

@iseeyou

iseeyou, 2021-02-24:
> 
> 
> 重新测试了一下, 发现直接用脚本start时还是用旧配置生效的, 因此重写了cbi.apply时的处理逻辑, 提交时马上生效. 详见[msylgj/OpenWrt_luci-app@061b9ef](https://github.com/msylgj/OpenWrt_luci-app/commit/061b9efbdb959a7bf25c0779edd1158c611771ff)
> 
> @iseeyou

非常感谢

WYC-2020, 2021-02-24:
改了后试了还是不行，我机器先被分配了192.168.2.206.然后我绑定 指定ip为192.168.2.110，然后应用，最后在电脑上执行ipconfig/renew 获取到的ip还是206，不是110，不知道是我哪里没操作对还是咋的 @msylgj @iseeyou

msylgj, 2021-02-24:
> 改了后试了还是不行，我机器先被分配了192.168.2.206.然后我绑定 指定ip为192.168.2.110，然后应用，最后在电脑上执行ipconfig/renew 获取到的ip还是206，不是110，不知道是我哪里没操作对还是咋的 @msylgj @iseeyou

你这根本就不是arp的作用…要分配的ip是固定的用静态dhcp，arp起不到保留和固定ip的作用 只是一个约束和永久的关系，一般应对arp广播风暴和WOL这种场景

WYC-2020, 2021-02-24:
@msylgj ok明白了

WYC-2020, 2021-02-24:
其实不用add 和change 用replace 就可以了@msylgj

iseeyou, 2021-02-25:
@msylgj 

编译固件后我看/etc/init.d/arpbind 没有执行权限，这是我编译的问题吗？
我是下载你的luci-app-arpbind包后上传到 /home/用户/newifid2/package/lean 然后设置了组用户名的rwxrwxr-x rwxrwxr-x 权限。

是不是有更方便的方法？直接每次 ./scripts/feeds update -a && ./scripts/feeds install -a 自动更新下载你的包呢。比如编辑 feeds.conf.default 加上去你的 https://github.com/msylgj/OpenWrt_luci-app/tree/main/lean/luci-app-arpbind 如何写呢？我加了语法老不对。

msylgj, 2021-02-25:
> @msylgj
>
> 编译固件后我看/etc/init.d/arpbind 没有执行权限，这是我编译的问题吗？
> 我是下载你的luci-app-arpbind包后上传到 /home/用户/newifid2/package/lean 然后设置了组用户名的rwxrwxr-x rwxrwxr-x 权限。
>
> 是不是有更方便的方法？直接每次 ./scripts/feeds update -a && [...]

我是直接svn co到package里面的 但是会独立有一个处理权限的chmod脚本跑一遍，具体看我repo的yml

iseeyou, 2021-02-25:
> 我是直接svn co到package里面的 但是会独立有一个处理权限的chmod脚本跑一遍，具体看我repo的yml

权限问题解决了，谢谢。