coolsnowwolf/lede #6386: 编译arm v8 机器失败报错

zzlinwq, 2021-02-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
/home/vibit/lede/staging_dir/host/bin/make_ext4fs -L rootfs -l 167772160 -b 4096 -m 0 -J -T 1613383373 /home/vibit/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/root.ext4 /home/vibit/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/
Creating filesystem with parameters:
    Size: 167772160
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 5120
    Inode size: 256
    Journal blocks: 0
    Label: rootfs
    Blocks: 40960
    Block groups: 2
    Reserved blocks: 0
    Reserved block group size: 15
error: ext4_allocate_best_fit_partial: failed to allocate 266 blocks, out of space?
/home/vibit/lede/include/image.mk:356: recipe for target '/home/vibit/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/root.ext4' failed
make[5]: *** [/home/vibit/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/root.ext4] Error 1
make[5]: Leaving directory '/home/vibit/lede/target/linux/armvirt/image'
Makefile:20: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/vibit/lede/target/linux/armvirt'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/vibit/lede/target/linux'
time: target/linux/install#987.67#137.69#1040.48
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/vibit/lede'
target/Makefile:19: recipe for target '/home/vibit/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install' failed
make[1]: *** [/home/vibit/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/vibit/lede'
/home/vibit/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
vibit@vultr:~/lede$ 
vibit@vultr:~/lede$ 

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
arm v8  贝壳云RK3328CPU 
A：

### (3) 详细日志
用美国VPS 编译
编译X86 64位固件正常
编译ARM V8报错
vibit@vultr:~/lede$ uname -a
Linux vultr.guest 4.15.0-132-generic #136-Ubuntu SMP Tue Jan 12 14:58:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 18系统
四核8G编译机

A：

camino13, 2021-02-17:
rootfs空间小了，要么增加空间，要么少选些插件

minlang112, 2021-02-17:
同楼上，你插件选太多了吧

closed 2021-06-17 by coolsnowwolf